SpacehuhnTech/esp8266_deauther #85: Can'

majf1999, 2017-03-05:


H1ghTech, 2017-03-05:
http://pastebin.com/RUv6XxS3

joneroy, 2017-03-05:
@majf1999 

Which board you you using:

https://github.com/spacehuhn/esp8266_deauther/blob/master/README.md

Ultimatefire54, 2017-03-05:
I was trying to upload to my esp01 last night and got the same error. In his post, within the error report it says esp generic 512kbs which is the esp01. It doesn't seem to work for the 01, and in the install spacehunh says comment the mac vendor list in data.h for esp01 but that doesn't work.

joneroy, 2017-03-05:
You read new readme

https://github.com/spacehuhn/esp8266_deauther/blob/master/README.md

Ultimatefire54, 2017-03-05:
I read it and it says the same thing about the esp01, 'comment the mac vendor list for esp01'

spacehuhn, 2017-03-06:
> The only thing you will need is a computer and an ESP8266 with at least 1Mb of flash memory.

> (If you using an ESP-01 with just 512kb, you can comment out the mac vendor list in data.h.)

so what is your problem with that? comment out the bytes and you're fine.

Ultimatefire54, 2017-03-06:
Commenting out any part of data.h results in compiler errors

N0vaPixel, 2017-03-07:
You don't have enough flash memory
`section .irom0.text' will not fit in regionirom0_0_seg`

I'll try later to reduce the size of the program.

spacehuhn, 2017-03-07:
@N0vaPixel you can't without deleting important parts. He needs to comment out a part of the mac vendor list. There is no way around this. the array is just to big.

N0vaPixel, 2017-03-07:
@spacehuhn I see, it may be a problem for many users, but the mac vendor list is very useful.

spacehuhn, 2017-03-07:
@majf1999 if you have a ESP-01 with 1mb (you can test this with the ESP8266 CheckFlashConfig example) you can set the flash size to 1mb with 64,128 or 256kb SPIFFS - just don't set it to 512kb
[image]

N0vaPixel, 2017-03-08:
#99

closed 2017-03-11 by spacehuhn