SpacehuhnTech/esp8266_deauther #86: Feature Request - Capture PSK handshakes

lamkeysing92, 2017-03-05:
Awesome work on the ESP8266 deauther. But I wonder is it possible to capture the 4 way handshake of WPA authentication? Not sure where would it be saved though. May be on the browser at runtime?

H1ghTech, 2017-03-05:
Read the readme!

spacehuhn, 2017-03-05:
> Can it sniff handshakes?

> The ESP8266 has a promiscuous mode in which you can sniff packets, but handshake packets are dropped and there is no other way to get them with the functions provided by the SDK.
Maybe someone will find a way around this barrier but I wasn't able to.

closed 2017-03-05 by spacehuhn

anudeepND, 2017-03-07:
It is possible but you can't see full conversation. You can't sniff all 4 packets

risinek, 2021-04-20:
ESP32 does not limit the frame buffer, so this can be done on ESP32. I created a [project for ESP32 deauth attack with handshake capture](https://github.com/risinek/esp32-wifi-penetration-tool) if you don't insists on ESP8266

SAQO101, 2024-04-11:
Bro can you make your project to work in cli ? Cos it works in browser only possible to make it to work in cli plssss 
I have been searching a lot but couldnt find anything except your project pls bro

SAQO101, 2024-04-11:
> ESP32 does not limit the frame buffer, so this can be done on ESP32. I created a [project for ESP32 deauth attack with handshake capture](https://github.com/risinek/esp32-wifi-penetration-tool) if you don't insists on ESP8266

Bro can you make your project to work in cli ? Cos it works in browser only possible to make it to work in cli plssss 
I have been searching a lot but couldnt find anything except your project pls bro