SpacehuhnTech/esp8266_deauther #87: Built-in LED Not Working Testing Branch

joneroy, 2017-03-05:
Built-in LED not working when attack is running i'm using testing branch

tpmodding, 2017-03-05:
kannst den mal nicht blocken oder so? @spacehuhn

N0vaPixel, 2017-03-05:
From pull request 
> Using built-in LED when an attack is running. (+ options)

Did you enabled it using the setting web page ?

H1ghTech, 2017-03-05:
Yes of course

N0vaPixel, 2017-03-05:
This is not happening to me after a fresh install from the testing branch, so try to reset your settings first

N0vaPixel, 2017-03-05:
Which board are you using.

joneroy, 2017-03-05:
@N0vaPixel 

> Did you enabled it using the setting web page ?

Yes

> This is not happening to me after a fresh install from the testing branch, so try to reset your settings first

i'm fresh install from the testing branch alos use reset settings

> Which board are you using ?

Tested NodeMcu board and wemos D1 mini

N0vaPixel, 2017-03-05:
@joneroy  @H1ghTech Can you past here the output of your serial monitor when you start an attack ?

joneroy, 2017-03-05:
@N0vaPixel 

settings:
SSID: pwned
SSID length: 5
SSID hidden: 0
password: deauther
password length: 8
channel: 1
Scan hidden APs: 1
deauth reson: 1
attack timeout: 300
attack packet rate: 10
client scan time: 10
attack SSID encrypted: 1
use built-in LED: 1
nameList cleared

starting...
starting WiFi AP
SSID: pwned
Password: deauther
generating Macs...done
getting settings json
{"ssid":"pwned","ssidHidden":0,"password":"deauther","apChannel":1,"apScanHidden":1,"deauthReason":1,"attackTimeout":300,"attackPacketRate":10,"clientScanTime":10,"attackEncrypted":1,"useLed":1,"nameList":[] }
done
getting AP scan result JSON { "aps":[ ] }
done
getting settings json
{"ssid":"pwned","ssidHidden":0,"password":"deauther","apChannel":1,"apScanHidden":1,"deauthReason":1,"attackTimeout":300,"attackPacketRate":10,"clientScanTime":10,"attackEncrypted":1,"useLed":1,"nameList":[] }
done
getting AP scan result JSON { "aps":[ ] }
done
starting AP scan...
MAC - Ch - RSSI - Encrypt. - SSID - Hidden
c8:3a:35:68:29:e8 - 3 - -35 - WPA2 - BREEZY - 0
00:90:4c:ec:47:a8 - 4 - -91 - none - ADYYTFlG - 0
dc:1a:c5:68:2f:e7 - 6 - -93 - none - vivo Y51L - 0
scan done
getting AP scan result JSON ...{ "aps":[ {"i":0,"c":3,"m":"c8:3a:35:68:29:e8","ss":"BREEZY","r":-35,"e":4,"se":0},{"i":1,"c":4,"m":"00:90:4c:ec:47:a8","ss":"ADYYTFlG","r":-91,"e":7,"se":0},{"i":2,"c":6,"m":"dc:1a:c5:68:2f:e7","ss":"vivo Y51L","r":-93,"e":7,"se":0}] }
done
{ "aps":[ {"i":0,"c":3,"m":"c8:3a:35:68:29:e8","ss":"BREEZY","r":-35,"e":4,"se":0},{"i":1,"c":4,"m":"00:90:4c:ec:47:a8","ss":"ADYYTFlG","r":-91,"e":7,"se":0},{"i":2,"c":6,"m":"dc:1a:c5:68:2f:e7","ss":"vivo Y51L","r":-93,"e":7,"se":0}] }
getting AP scan result JSON ...{ "aps":[ {"i":0,"c":3,"m":"c8:3a:35:68:29:e8","ss":"BREEZY","r":-35,"e":4,"se":0},{"i":1,"c":4,"m":"00:90:4c:ec:47:a8","ss":"ADYYTFlG","r":-91,"e":7,"se":0},{"i":2,"c":6,"m":"dc:1a:c5:68:2f:e7","ss":"vivo Y51L","r":-93,"e":7,"se":0}] }
done
seect 0 - 1
getting AP scan result JSON ...{ "aps":[ {"i":0,"c":3,"m":"c8:3a:35:68:29:e8","ss":"BREEZY","r":-35,"e":4,"se":1},{"i":1,"c":4,"m":"00:90:4c:ec:47:a8","ss":"ADYYTFlG","r":-91,"e":7,"se":0},{"i":2,"c":6,"m":"dc:1a:c5:68:2f:e7","ss":"vivo Y51L","r":-93,"e":7,"se":0}] }
done
getting client scan result JSON { "clients":[] }
done

starting client scan
set channel to 3
changing to channel 4
changing to channel 6
stopping client scan after 10000ms
0: 375 Universa e0:2a:82:fb:48:08
starting WiFi AP
SSID: pwned
Password: deauther
getting client scan result JSON { "clients":[{"i":0,"p":375,"m":"e0:2a:82:fb:48:08","n":"","v":"Universa","s":0}] }
done
getting client scan result JSON { "clients":[{"i":0,"p":375,"m":"e0:2a:82:fb:48:08","n":"","v":"Universa","s":1}] }
done
getting attacks JSON...{ "aps": ["BREEZY"], "clients": ["e0:2a:82:fb:48:08 Universa - "], "attacks": [{"name": "deauth","status": "ready","running": 0},{"name": "beacon (clone)","status": "ready","running": 0},{"name": "beacon (list)","status": "no SSID","running": 0}],"ssid": []}
done
0
0
1
1
1
[... 538 lines cut ...]
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done
running deauth attack done

N0vaPixel, 2017-03-05:
Past big datas here: [http://pastebin.com/](url)

Mmmh, interesting.... The LED function was called.
Make a new sketch and update the firmware of your board with this code: [http://pastebin.com/Y9cJt1z8](url)

This is a blink test with the built-in LED

joneroy, 2017-03-05:
@N0vaPixel 

> Blink test with the built-in LED

LED Blink Not Working

N0vaPixel, 2017-03-05:
Make sure you have a LED on your board, because she seems to be death 💀

joneroy, 2017-03-05:
@N0vaPixel 

# **When i'm uploading the sketch it's built-in LED Blink**

joneroy, 2017-03-05:
@N0vaPixel 

It's not my prob @H1ghTech

N0vaPixel, 2017-03-05:
> LED Blink Not Working

> When i'm uploading the sketch it's built-in LED Blink

It's difficule to me to understad you000

joneroy, 2017-03-05:
> LED Blink Not Working after done upload sketch

> When i'm uploading the sketch it's built-in LED Blink

N0vaPixel, 2017-03-05:
This is not the same LED, this is the RX LED not the IN-BUILD LED

joneroy, 2017-03-05:
@N0vaPixel 

[image]

this led

N0vaPixel, 2017-03-05:
If this LED blink during the upload of a sketch, this is not the built-in LED.
Look at the arduino uno, there is 3 LEDs : TX, RX and BUILT-IN LED (connected to i/o 13)
The build-in led must be connected to the inputs/ouputs of the board.

joneroy, 2017-03-06:
@N0vaPixel 

Thank You

N0vaPixel, 2017-03-06:
@joneroy  Sorry but you cannot use this feature with this board, i think you don't have a built-in LED.

joneroy, 2017-03-06:
@N0vaPixel 

You right.

This LED Blink feature with RX LED it's possible

N0vaPixel, 2017-03-06:
Wow, viewed some documentation, the LED of the esp is connected to GPIO2
I think you don't have a called "built-in led", but you can use the esp one.

I'll commit this change, the program will use the LED of the esp instead of the one on the board to increase compatibility.

nisc3, 2017-03-06:
This led should blink right?
http://i.imgur.com/fehC28h.jpg

joneroy, 2017-03-06:
> This led should blink right?

This is RX LED on board BUILT-IN LED

## **With on board BUILT-IN LED**

1.nodemcu amica

[image]

## **Without on board BUILT-IN LED**

2.nodemcu lolin

[image]

3.wemos D1 mini

[image]

nisc3, 2017-03-06:
But RX LED on my board is not blinking

joneroy, 2017-03-06:
@nisc3 

Read full conversation i think you found somthing helpful

spacehuhn, 2017-03-06:
Works on my new nodeMCU! (it has a LED yey! 🙌 )
But not on the "nodeMCU v3 lolin board", also not on my ESP-01 and the wemos d1 mini pro.

N0vaPixel, 2017-03-06:
Since many boards dind't have a built-in LED, i'll update the software to use the ESP one, connected to gpio 2, and removed the incompatible (in most cases) constant BUILTIN_LED.

joneroy, 2017-03-06:
@N0vaPixel 

Most of the esp8266 board not available "BUILTIN_LED" feature is good but board compatibility main issue
Thank you.

joneroy, 2017-03-06:
@N0vaPixel 

Now LED Working!!

N0vaPixel, 2017-03-06:
@joneroy  I've never designed the program to make the LED blink

joneroy, 2017-03-06:
@N0vaPixel 

No Prob

N0vaPixel, 2017-03-06:
Perfect !
Thank you for your help 💪

N0vaPixel, 2017-03-06:
Why did you reopened it ?

joneroy, 2017-03-06:
@N0vaPixel 

Sorry i mistakenly press button.

closed 2017-03-06 by joneroy

anudeepND, 2017-03-07:
An option to choose the behaviour of LED like blinking or constant would be great.

N0vaPixel, 2017-03-07:
@anudeepND Thank you for helping us. But sorry it's useless. Memory is limited and we can't add extra bytes for funny features

anudeepND, 2017-03-08:
Ok I understand

anudeepND, 2017-03-08:
@N0vaPixel did you tried MAC changer? All my ESPs are busy, so if possible try it out