SpacehuhnTech/esp8266_deauther #91: doesnt work

5465da, 2017-03-06:
I uploaded the code to my esp8266 and run it on my phone at 192.168.4.1. And tried to deauth my client on the wifi and it doesn't work. It doesn't disconnect the client

H1ghTech, 2017-03-06:
Do u have a 5ghz network?

5465da, 2017-03-06:
Yes,It is connected to my wifi extender and I tested on my wifi extender 2.4ghz

5465da, 2017-03-06:
Plus my connection to the esp8266 keep disconnecting

tobozo, 2017-03-06:
ESP8266 can't do 5Ghz, and 5Ghz AP are usually protected against deauth attacks.

5465da, 2017-03-06:
But I tested it on 2.4ghz... The configuration on my router 2.4ghz and 5ghz is enabled

5465da, 2017-03-06:
Nvm tmr I will test it again

tobozo, 2017-03-06:
Just edited my comment, make sure your router has no anti DDOS enabled when you test. 
Also deauth packets won't affect devices using the 5Ghz band.

spacehuhn, 2017-03-06:
> Deauth attack won't work

> If you see 0 pkts/s on the website you have made a mistake. Check if you have followed the the installation steps correctly and that the right SDK installed, it must be version 2.0.0!
If it can send packets but your target don't loose its connection then the WiFi router uses 802.11w and it's protected against such attacks or they communicate via 5 GHz WiFi, which the ESP8266 doesn't support.

5465da, 2017-03-07:
I encountered another problem when I scan for clients, the esp8266 will disconnect to the phone connected to it.

5465da, 2017-03-07:
`
[image]
`

5465da, 2017-03-07:
[image]

spacehuhn, 2017-03-07:
read the readme

closed 2017-03-07 by spacehuhn