SpacehuhnTech/esp8266_deauther #93: Seems like new program isn't loading on the board.

whyAllUsernamesAreTaken, 2017-03-06:
I uploaded the new program to the board using the previous settings which worked fine with the previous program but now i don't get any AP. 
All the previous programs are working fine but not this one. 
The ssidHidden is also false and i have not touched any line of the program.
I get the full signal in other network AP but when i try to connect to it by giving the ssid of my ap  it does not connects. I have also tried giving it the previous ssid that i have assigned manually by settings webpage.
**Please help.**

joneroy, 2017-03-06:
@whyAllUsernamesAreTaken 

you are using sdk_fix new update

spacehuhn, 2017-03-07:
can you test this with other boards? Is the old version still working? Have you tested one of the example code to see if it's a hardware problem...?

whyAllUsernamesAreTaken, 2017-03-07:
The old versions are working fine. I have also tested the example code and it is also working. and i am using sdk_fix updated that you suggested.

whyAllUsernamesAreTaken, 2017-03-07:
can anyone please help me?

N0vaPixel, 2017-03-07:
It's difficult to find the problem. First check the version of your sdk, are you using the version 2.0.0 ?

whyAllUsernamesAreTaken, 2017-03-07:
Yes i am using 2.0.0

N0vaPixel, 2017-03-07:
Try the example of your sdk: `\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\examples\WiFiScan`

whyAllUsernamesAreTaken, 2017-03-07:
I have already tried some of the examples they are working fine, but when i upload the deauther program firstly i cannot find the ap and i get the full signal in other network which is hidden but when i try to connect it fails.

spacehuhn, 2017-03-07:
If debug is set to true, you should see all information you need in the serial monitor, the SSID, the password and if it's hidden.

whyAllUsernamesAreTaken, 2017-03-07:
Ok i will share the debug report after checking, thanks.

whyAllUsernamesAreTaken, 2017-03-08:
The  language in the serial monitor is not understandable.

N0vaPixel, 2017-03-08:
Set the baud rate to 115200

whyAllUsernamesAreTaken, 2017-03-08:
yes it is 115200, should i try with 9600?

N0vaPixel, 2017-03-08:
Normaly not. The unreadable text is contained on one or multiple lines ?

whyAllUsernamesAreTaken, 2017-03-08:
when i used the 9600 baud rate the symbols on serial monitor keeps on increasing and the led on the board is on. What to do next ?
[image]

spacehuhn, 2017-03-08:
Set the baud rate to 115200 and show us the log.

whyAllUsernamesAreTaken, 2017-03-08:
[image]

N0vaPixel, 2017-03-08:
[image]

N0vaPixel, 2017-03-08:
And you restarted the esp ?

whyAllUsernamesAreTaken, 2017-03-08:
yes now i got it. But i am not getting the ap while i am scanning
andthe ap is not named as pwned.
[image]

N0vaPixel, 2017-03-08:
The AP while scanning ? You are trying to connect to your esp AP ?

whyAllUsernamesAreTaken, 2017-03-08:
The ap is hidden but i have not set the ssidHidden to true how is that possible ?
I have connected to the ap it was hidden.

whyAllUsernamesAreTaken, 2017-03-08:
I was not getting my esp AP. It was hidden by default but i don't know how?

N0vaPixel, 2017-03-08:
Just understood ^^ When you said scan for AP i thought this was to find target AP

N0vaPixel, 2017-03-08:
You may have downloaded an older version of the soft. Because the settings structure in eeprom memory has changed, the old one may have Interfered the current soft

whyAllUsernamesAreTaken, 2017-03-08:
How can i solve this brother?

N0vaPixel, 2017-03-08:
do you know how to connect to hidden networks ?

whyAllUsernamesAreTaken, 2017-03-08:
Yes. Well i can change the setting on the browser for my esp.

whyAllUsernamesAreTaken, 2017-03-08:
Now you can remove the bug label from this issue as the problem is solved, Thanks again brother.

N0vaPixel, 2017-03-08:
Good news. Now you can try all the features, see how the soft works ect... Have fun 👍

closed 2017-03-08 by N0vaPixel