SpacehuhnTech/esp8266_deauther #101: iOS 9.3 issue

benjenq, 2017-03-07:
I've tested it on NodeMCU and and it works! fantastic (via PC). A little bit issue need to report.

 it doesn't work on iOS 9.3 device (can not do scan action via iOS 9.3 device). iOS 10 is OK.

I found it's javascript issue and need to fix if you want to play via iOS 9 device:

**functions.js : line 1**
```javascript
function getResponse(adr, callback, timeoutCallback = function(){location.reload()}, timeout = 8000){
```
need to change as following:
```javascript
function getResponse(adr, callback, timeoutCallback, timeout){
	if(timeoutCallback === undefined) { 
		 timeoutCallback = function(){
		 	location.reload()
		 };
	}
	if(timeout === undefined) { 
		timeout = 8000; 
	}
```

And 
**clients.html: line 75**
```javascript
function getResults(reload = false){
```

need to change as following:
```javascript
function getResults(reload){
  if(reload === undefined){
     reload = false;
  }
```

N0vaPixel, 2017-03-07:
@benjenq  Thanks for you help !
This means that old browser can't handle default parameters ?

benjenq, 2017-03-08:
It throw out a error in iOS 9.3 : 
'SyntaxError: Unexpected token '='. Expected a ')' or a ',' after a parameter declaration.'

It seem like old version of Apple safari issue.

N0vaPixel, 2017-03-08:
The changes have been committed (testing branch), it should be fixed. Thanks. 👍

closed 2017-03-08 by N0vaPixel