SpacehuhnTech/esp8266_deauther #98: Spam WOL packets

Hammad1029, 2017-03-07:
Just like spamming deauth packets, is it possible to spam WOL(wake on lan) packets? add it please # #

joneroy, 2017-03-07:
@Hammad1029 

It's complicated feature "Spam WOL packets"

spacehuhn, 2017-03-07:
How should you send wake on lan packets without knowing the mac of the target?

Hammad1029, 2017-03-07:
by scanning the ap? scanning the ap currently and going on the clients does reveal the client address.

spacehuhn, 2017-03-07:
yes, avtive devices! ...devices which doens't care about WOL packets you send them.

spacehuhn, 2017-03-07:
You can't scan for sleeping devices. You'd need to preset the MACs and at this point you'd probably have physical access and don't need to send these packets. I don't see the use for this project.

N0vaPixel, 2017-03-08:
You're probably talking about "wake on wireless lan"
+ Like @spacehuhn said you need to know your targets
+ WOL packets are often protected
+ Like WOL, to use WOWLAN technology you must be connected to an AP.
Sorry but we can't do anything
Thanks for helping us to find new ideas :)

closed 2017-03-08 by N0vaPixel