SpacehuhnTech/esp8266_deauther #99: .bin Releases for flashing with esp8266flasher

tpmodding, 2017-03-07:
Hey guys,

some users are to lazy/stupid to read the readme then they are starting to spam with issues...
So here is a solution for them:
you need two things:
1. ESP8266Flasher
2.the esp8266deauther precompiled binarys

Quick Tut:
I use this with NodeMCU and this settings:
[image]

If you click upload you can see the path to the .bin where the flash starts...
[image]

then start esp8266flasher and select the .bin
[image]

click flash and wait :)
[image]

Here is a actuall bin from testing branch, if you want to test...

[esp8266_deauther-testing.bin.zip](https://github.com/spacehuhn/esp8266_deauther/files/825371/esp8266_deauther-testing.bin.zip)

you could tag them with a version number and upload it at as release here
https://github.com/spacehuhn/esp8266_deauther/releases

greetings TP

spacehuhn, 2017-03-07:
Yes I think we're at a point where this is badly needed! Thanks for the advice.

spacehuhn, 2017-03-07:
Also solves the 512kb problem. I just create 2 bin files.

Splodgie, 2017-03-08:
Last comment, Goodbye.

N0vaPixel, 2017-03-08:
@tpmodding Nice flashing alternative ! Thx

> Last comment, Goodbye.

#32 
Love theres guys like @Splodgie who think that we are their employees. We don't owe you anything.
It's time for you to make things yourself 😄  
So goodbye too

spacehuhn, 2017-03-08:
WTF I don't understand these people. He deleted every of his posts. Whole conversations here doesn't make sense anymore. This is ridiculous.

chinahk, 2017-03-08:
saying people lazy stupid for asking question is insult he help me very muc to get program load working he send me here helpme all time.

spacehuhn, 2017-03-08:
Nobody said he is stupid. This gone a bit off topic here, I think we can close this.

closed 2017-03-08 by spacehuhn

FirstPure, 2017-03-08:
What a bummer If it was not for Splodgie I would have gave up ages ago, I watched a indian YT video got no help until he offered to walk me through it god did I ask he some questions must have been 50 or more. 
I came here and found he's gone I wonder why, tpmodding  lazy/stupid, N0vaPixel links his post #32  We don't owe you anything.  WTF I don't understand these people. you dont understand anybody but your self.
(at least I have his EMail).
Why dont you make this private so none of us stupid peiople bother you.

tpmodding, 2017-03-08:
if you have problems, i help you, i show it you, i can explain it you, and if nothing helps, i can connect to your pc via teamviewer and help, if you have problems....

but most of the issues was " 'wifi_send_pkt_freedom' was not declared" that means that those people ar to lazy to do anything and read....the readme shows you step by step what to do...if you follow those steps you DONT get this error...if you have other problems, you are welcome to ask for support...don't get me wrong...i'm even not a coder...but i am able to read first everything, try it, if it dont work, to search in the issues or google, and if then nothing works or ask for support...

they added a new feature, which was explained in the readme what you have to do after those changes...but nobody reads them and say "new testing branch dont work"....no infos why, what he has done, which hardware nothing... and my message was not for a special user or something else...

FirstPure, 2017-03-08:
I think thats called Back Peddling M8.

tpmodding, 2017-03-08:
oh no, sure not! its still a solution for lazy/stupid ones who dont want to read the readme and open issues! but enough said now...

spacehuhn, 2017-03-08:
@FirstPure you, like many others, miss the point that this is no forum. no arduino forum and no espressif support. And this project is not about explaining everyone on this world how the basics of microcontrollers work. We help as much as we can, but most people don't even read the readme correctly or search for similar issues. Also most poeple don't even give us the information needed to get or solve their problem. We don't need to help you, we use our free time to do this project and don't get paid for this! And to comment things like this on an issue where we are clearly helping to make the installation process easier for everyone is very ignorant.

I get that you 're offended by @tpmodding statement, but he has his point. This is not what open source projects and github is for. Get in our position first before blaming us. We are providing more help and a way better readme as most projects. Sorry that it's to complicated for you.