SpacehuhnTech/esp8266_deauther #102: Wifi ProbeRequest

GraafG, 2017-03-09:
Is it posible to expand this project to include ProbeReequests?
This will help spam wifi trackers in cities.
A mac-list from SD is even better.

The basic code can be found here:
https://github.com/GraafG/WifiProbeRequest

spacehuhn, 2017-03-09:
So that's just to confuse/spam trackers?

GraafG, 2017-03-09:
Yes, in The Netherlands wifitracking is used in public spaces and also in stores.
I want to put out random probe requests to random SSID's with random MAC's from random MAC's.
A better solution would be to use a list with allowed SSID's and MAC's in/out to use only mobile vendors for sender MAC.

I asked the company bluemark what they are using for this tracking, they said they are using:
proberequests and data null frames.

https://bluemark.io/
http://bluetrace.eu/
http://www.acurity.nl/klantentelling/wifi-tracking/
http://crosscan.com/de/
http://citytraffic.nl/
https://bluemark.io/

spacehuhn, 2017-03-09:
Ok a SSID list is already there so we could use that and the MACs are also generated random and change every few seconds in the attacks. So it's just a matter of replacing the beacon packet with a probe request packet.

Could be very usefull! Thanks for the infos and links.

spacehuhn, 2017-03-09:
Maybe it would also help to change the TX power randomly on this attack to make it harder to track you as spammer?

spacehuhn, 2017-03-10:
141c9f17a30658ff78b111c6772496bf40d3508e

There you go :)

closed 2017-03-11 by spacehuhn