SpacehuhnTech/esp8266_deauther #103: Can't upload the sketch

dankocean, 2017-03-09:
Hey, so I'm having this issue with the sketch upload. First, how do I know what is my ESP8266 board? Second, when I'm trying to do the upload errors occur. The errors say that there are no such files or directories like # include data.h, Mac.h and Namelist.h.

tpmodding, 2017-03-09:
did you read this one?
https://github.com/spacehuhn/esp8266_deauther/blob/master/README.md

dont you know what you bought? did you try to search on google?
https://blog.squix.org/2015/03/esp8266-module-comparison-esp-01-esp-05.html

spacehuhn, 2017-03-09:
looks like some kind of directory error. Arduino don't find the files, I can't say why though, I'm not in your computer.

dankocean, 2017-03-09:
Yeah, I've read it. I have this one NodeMcu Lua http://www.ebay.com/itm/171907569391?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

I think I've followed your instructions carefully. But the errors can't occur if you pick the wrong board and the right port, right...? :D

spacehuhn, 2017-03-09:
> But the errors can't occur if you pick the wrong board and the right port, right...? :D

I would not think so. It's really hard to say what the problem is here, probably something with Arduino or your copy of this project. Maybe you go with the easier method by using the nodemcu flasher and uploading the bin file?

dankocean, 2017-03-09:
Now that I remember, there was this one part of the project that got me confused. When I pasted  ESP8266WiFi.cpp and ESP8266WiFi.h from the SDK_fix to src, the files were already found in there, so I overwrote them.

spacehuhn, 2017-03-09:
That's necessary yes.

FirstPure, 2017-03-10:
There just F'ing you about M8. 

Click on this link https://github.com/spacehuhn/esp8266_deauther/releases/download/1.0/esp8266_deauther_1.0_1mb.bin Flash it to your ESP Using ESP8266Flasher https://onedrive.live.com/?authkey=%21AMHy3KM8NWX7yFI&cid=36E983D4FED6FE4F&id=36E983D4FED6FE4F%218572&parId=36E983D4FED6FE4F%218557&action=locate Job Done.

spacehuhn, 2017-03-10:
https://github.com/spacehuhn/esp8266_deauther#uploading-the-bin-files

### Uploading the bin files  

**Note:** the 512kb version won't have the full MAC vendor list.  

**0** Download the current release from [here](https://github.com/spacehuhn/esp8266_deauther/releases)  

**1** Upload using the ESP8266 flash tool of your choice. I recommend using the [nodemcu-flasher](https://github.com/nodemcu/nodemcu-flasher). Alternatively you can use the official [esptool](https://github.com/espressif/esptool) from espressif.

FirstPure, 2017-03-10:
> https://github.com/spacehuhn/esp8266_deauther#uploading-the-bin-files
>
> [https://avatars2.githubusercontent.com/u/3826044?v=3&s=400]<https://github.com/spacehuhn/esp8266_deauther#uploading-the-bin-files>
>
> GitHub - spacehuhn/esp8266_deauther: ESP8266 deauther<https://github.com/spacehuhn/esp8266_deauther#uploading-the-bin-files>
> github.com
> README.md ESP8266 Deauther. Build your own WiFi jammer with an ESP8266. Contents. Introduction. What it is; How it works; What an ESP8266 is; How to protect against it

I have been in touch with Sploge he gave me his email because i was inneed of a lot of help

He was working on 5 youtube videos to help people working with the arduino ide and ESP covering all the install and program methods to get this project working, He read the post and found it insulting to those he wished to help. he did not wish to be connected to this arrogance attitude towards others so removed himself and posts related to your project. after reading the replies after he left he said he is sure he made the correct decision for him he wishes you well with your project.

TFP

closed 2017-03-18 by spacehuhn