SpacehuhnTech/esp8266_deauther #104: Not deauthing

FrostyModz, 2017-03-10:
Devices still have internet access. I can only create fake APs (beacon)

poslogic, 2017-03-10:
Please read and follow the readme step by step.

Then if you still have problems you will need to provide a lot more information than you have already before anyone could try and help you.

spacehuhn, 2017-03-10:
**Deauth attack won't work**

If you see 0 pkts/s on the website you have made a mistake. Check if you have followed the the installation steps correctly and that the right SDK installed, it must be version 2.0.0!  
If it can send packets but your target don't loose its connection then the WiFi router uses [802.11w](#how-to-protect-against-it) and it's protected against such attacks or they communicate via 5 GHz WiFi, which the ESP8266 doesn't support.

Mcewen123, 2017-03-10:
I have 4 laptops and it doesn´t work of any of the(yeeee maybe 1 in 20 tries it works) but if I connect the phone to the wifi the deauther works every time

tpmodding, 2017-03-10:
may your laptops are connected over 5ghz

Mcewen123, 2017-03-10:
No, have a HP 8760w(http://h20564.www2.hp.com/hpsc/swd/public/detail?swItemId=ob_93396_3#tab3), 8560w, 8770w and it can´t deauther any of them

spacehuhn, 2017-03-10:
Do you find your laptops when you scan for clients in your network?

Mcewen123, 2017-03-10:
Yes.........
I have a Toshiba laptop, that I can deauther if I use the client but not the other ones

spacehuhn, 2017-03-10:
Well then be happy that your other devices are protected against this attack :)

The only way to be sure is to test it with other tools like aircrack or mdk3.

ChaudhryAbdulAziz47, 2017-03-13:
Same Here Bro! @spacehuhn I've tried to run it on multiple laptop's! including 802.11 b/g/n router and laptops with different network cards; same problem exists! It tells me de-authenticating between 9 and 10 pkt/s but internet on laptops keep working just fine. I tried with aircrack-ng and also with mdk3; it takes down the whole network within seconds! But with ESP8266_deauther! It scans and networks and clients just fine but once it comes to deauth attack it stays on 9 to 10pkt/s with no affect on victims machine whatsoever. I look forward to hear from you soon!

spacehuhn, 2017-03-13:
OK. Can you check with wireshark if the deauth packets from the esp8266 are valid and get to the clients?

FrostyModz, 2017-03-13:
In stead of 10 packets use something like 100

ChaudhryAbdulAziz47, 2017-03-13:
Thanks @spacehuhn and @FrostyModz your help is much appreciated; It's finally working! I checked the deauth packets from mdk3 and also from esp8266! with 10pkt/s i was able to take down devices such as iphone and my android phone. But for laptops you'll need to bump up the speed of packet flood to 100pkt/s. (Thanks once again @FrostyModz ), Now the only problem left is that clients only appear if you scan wireless networks couple of time. I scanned wireless networks and selected my network; clients didn't appear until it was my 3rd time i restarted the device and selected my network and scanned for clients.

Mcewen123, 2017-03-14:
> In stead of 10 packets use something like 100

> 

Thanks @FrostyModz works like a charm

closed 2017-03-18 by spacehuhn

ddatt12, 2017-04-24:
Thanks, how do I increase the amount of packets sent though?

Mcewen123, 2017-04-24:
Go to the settings page and : 
[image]