SpacehuhnTech/esp8266_deauther #106: [Version Request] ESP32 implementation

tobozo, 2017-03-10:
Seems like [ESP8266 running complex apps over networking are in  danger of transient global heap underrun](https://github.com/warmcat/libwebsockets/issues/776) (sounds scary and I have no idea what this means).

In the comments, they say the ESP32 has enough memory to espace this problem. What is the problem with the ESP32 SDK (apart from the cryptic post saying only ARP can be fiddled with)? 
Is there a workaround like there is with SKD 2.0.0 / ESP8266?

spacehuhn, 2017-03-10:
No that's the problem. Have you read the forum post I attached to the links in the readme?

http://esp32.com/viewtopic.php?f=13&t=586&p=2648&hilit=wifi_send_pkt_freedom#p2648

They completely removed the `wifi_send_pkt_freedom` function from the esp32 SDK and without that you can't inject packets. And the new `esp_wifi_internal` function can't be used for that.

I couldn't find other functions for that.

spacehuhn, 2017-03-10:
Also very informative: https://esp32.com/viewtopic.php?p=3053#p3053

tobozo, 2017-03-10:
Ok I guess we can't just recycle the code from the ESP8266, but does it mean ESP32 cannot do packet injection (hardware wise), or does it just mean Espressif won't provide any code that would allow it?

Also I don't understand the posture, denying free packets but allowing ARP spoofing sounds a bit schizophrenic. If this is a lost cause I'll just close the issue and yell at my ESP32's.

spacehuhn, 2017-03-10:
> or does it just mean Espressif won't provide any code that would allow it?

yes exectly! You'd have to rewrite their code which is not open source :(

>  denying free packets but allowing ARP spoofing sounds a bit schizophrenic

I thought that too, but as it turns out it's not easy. The `esp_wifi_internal` function is really internal only and get's super buggy when using it, but in theory it could send (for example) ARP packets.

>  If this is a lost cause I'll just close the issue and yell at my ESP32's.

I do that too from time to time... 😄

closed 2017-03-11 by spacehuhn

dwillmore, 2018-08-16:
Does [this](https://github.com/espressif/esp-idf/blob/master/components/lwip/port/netif/wlanif.c#L105) change the ability of the ESP32 to support deauther?

spacehuhn, 2021-12-12:
Hi! There are many issues on this repo about ESP32 support, so I published a blog post to bring some clarity to this topic: https://blog.spacehuhn.com/esp32-deauther/ 
TL;DR: It's made for the ESP8266, and I have no plans to port this application to the ESP32. But it's open-source, so you're free to do it yourself if you want. I appreciate your understanding.