SpacehuhnTech/esp8266_deauther #107: does it support ap channel change so it can start the attack automatic  after channel change ????   ﻿

minanagehsalalma, 2017-03-11:


N0vaPixel, 2017-03-11:
Channel change is for the AP created by the esp.

minanagehsalalma, 2017-03-11:
@N0vaPixel no channel change is for the first wifi router...  and i meant if the script support channels hopping

spacehuhn, 2017-03-11:
Next time please ask your question clearly with a short title and a understandable description, thanks.

If the AP changes the channel while you're attacking it, you'd have to scan again to get the new channel (there's no other way).
If you have selected multiple APs which run on different channels it will automatically switch through them.

closed 2017-03-11 by spacehuhn

minanagehsalalma, 2017-03-11:
@spacehuhn    (
If you have selected multiple APs which run on different channels it will automatically switch through them.)  even if one of the APs changed has channel

spacehuhn, 2017-03-11:
no. you have to scan to get the new channel.

minanagehsalalma, 2017-03-11:
@spacehuhn  can mdk3 run on esp8266 nodemcu ????

anudeepND, 2017-03-12:
Nope.

minanagehsalalma, 2017-03-12:
@anudeepND  why not   ??

anudeepND, 2017-03-12:
Considering the memory and storage availability on esp it is not possible. And this project has all the best features. If you are looking to install mdk3 and other wireless tools out there, a $5 raspi zero is your best option.

minanagehsalalma, 2017-03-12:
@anudeepND can this version of mdk3  https://github.com/wi-fi-analyzer/mdk3-master run on the $5 raspi zero ???  do you have a video link about  how to use it ??

tpmodding, 2017-03-12:
yes
http://bit.ly/2ncxXBy

minanagehsalalma, 2017-03-12:
@tpmodding i didn't find a video !!!!!

N0vaPixel, 2017-03-12:
Impossible to run an os based programs (like mdk3) on the esp's microcontroller.
Aircrack is a dangerous tool for other users. So if you didn't know how it basicly works and how properly install it, you didn't deserve to use it, because your purpose will only be to annoy others. So please don't be like all these "script kiddies".
Mdk3 and many tools can run on the $5 raspi zero.

**I lock this issue.**