SpacehuhnTech/esp8266_deauther #110: can you add channel  hopping

minanagehsalalma, 2017-03-11:
i want the script to hop channels like mdk3..  cause of ap channel change...  can you do this  ??? please

spacehuhn, 2017-03-12:
there you go: 25a9dfe73c847b2585b407241a56c7b78f0ad3a9

closed 2017-03-12 by spacehuhn

minanagehsalalma, 2017-03-12:
@spacehuhn  does it hops all the 13 channels ????

N0vaPixel, 2017-03-12:
@minanagehsalalma  Look at the code, if channel hopping is enabled, the esp will send deauth packet for specified client/AP to all channels. Only channels 1 to 11 (USA band)

minanagehsalalma, 2017-03-12:
@N0vaPixel @spacehuhn  can you add channels 12 &  13 cause i am not from usa ??  please

N0vaPixel, 2017-03-12:
Me too i live in EU
I plan to add this feature, in settings you'll be able to choose your location.
USA (channel 1 to 11)
EU (1 to 13)
Japan (1 to 14)
I don't know the others

minanagehsalalma, 2017-03-12:
@N0vaPixel great..  add this option ..  the others are  not important ..

spacehuhn, 2017-03-12:
@minanagehsalalma how about doing it yourself? You don't need this feature anyway in 99% of the time.

N0vaPixel, 2017-03-12:
@minanagehsalalma  This is not my priority. All the wifi equipements in EU are limited to USA channels.

spacehuhn, 2017-03-12:
@N0vaPixel no need to add this. If someone really needs this, it can be easily changed by going into the code and replacing the 11 int the for loop with a 13 or 14. Like you said, most devices are limited to the USA channels anyway and the ESP's memory is limited!

minanagehsalalma, 2017-03-12:
@spacehuhn  which one   ???

minanagehsalalma, 2017-03-13:
@spacehuhn @N0vaPixel  in which code file

minanagehsalalma, 2017-03-18:
@spacehuhn @N0vaPixel I don't have the skills to do that...  can you explain more ??