SpacehuhnTech/esp8266_deauther #115: Beacon spam stops after a short period of time.

DGSE, 2017-03-13:
I was messing around with the latest release today and notice that the beacon spam feature was stopping itself after a minute or so. I'm not sure if this is intentional and I haven't noticed it, but it's quite annoying when trying to use that feature. 

=======================

To clarify further, when it would stop I was not currently controlling the ESP8266. I would turn it on, start a beacon spam, and let it sit for a minute. After a short period of time the ESP8266 will stop the spam and the LED indicator will shut off. I'm not sure if it'll do the same if I was using it and not just setting it down and walking away (metaphorically speaking, I was actually right next to it, just "AFK.") Like I said, this might be intentional and I may have looked over it. If that's the case please disregard this post.

H1ghTech, 2017-03-13:
I think the esp is running out of memory try setting the pkts/s down

N0vaPixel, 2017-03-13:
Hi @DGSE. There is nothing like that. The beacon spam should work as long as it stay enabled. Can you enable the monitor and let it run for a while, you may be able to see what's happened.

DGSE, 2017-03-13:
Sure thing @N0vaPixel, I'll try to do that as soon as I get off work tomorrow. I'd also like to add that I tried this on 3 different boards and all yielded the same result.

DGSE, 2017-03-13:
Update: Just timed it with a stopwatch, after multiple tests for accuracy, the beacon spam shuts itself off right before 5 minutes and 30 seconds.

spacehuhn, 2017-03-13:
Go to settings and set attack timeout to 0

closed 2017-03-13 by spacehuhn

DGSE, 2017-03-13:
Ah, so I did overlook it. Thank you @spacehuhn.

spacehuhn, 2017-03-13:
no Problem ;)