SpacehuhnTech/esp8266_deauther #116: how to translate language?

ErfanDL, 2017-03-13:
Hi. how can I translate web ui to other languages ? I trying edit html files but translation not changes when uploading to esp

thanks

N0vaPixel, 2017-03-13:
You need to convert into bytes the html files and update data.h
All informations are included in the data.h file

ErfanDL, 2017-03-13:
thanks for replay. can you tell me a example ? Im noob :D

N0vaPixel, 2017-03-13:
Edit your html file, copy/past the content into the minifier (minifier.html), click on button "minify + byte-ify", get the result and past it into the appropriate PROGMEM variable in data.h.

ErfanDL, 2017-03-13:
thanks for replay. is minifier.html program? or online tools ?

N0vaPixel, 2017-03-13:
Both, you can use any online tool that convert text into bytes.

spacehuhn, 2017-03-13:
@ErfanDL everything is in the htmlfiles folder, including a readme.

poslogic, 2017-03-13:
Open the minified.html file that is in the html folder of this code.

Then paste your version of new html into the top part.

Then click minify + bitify.

Then copy the bottom part into the data.h file between the {} brackets of the page you are changing.

Then save data.h

Then compile and flash to ESP8266

N0vaPixel, 2017-03-13:
Everything has been explained, i close this issue.

closed 2017-03-13 by N0vaPixel

ErfanDL, 2017-03-14:
thanks very much. my problem solved