SpacehuhnTech/esp8266_deauther #117: error 404 not found while building

ninoosewoudt, 2017-03-13:
Hello,

i first installed the bin but now i want to work on this code base myself but i encountered a problem.
I saw some one had the same problem as me.
---
I'm installing it on my nodeMCU and I'm facing some issues.
First when I'm building it with Arduino IDE I got some errors like:
load404 not defined, I had to move all prototypes to the top of the ino file.
---

but i cant find out which prototypes to move and to where.
i would apreciate some help.

thnx.

N0vaPixel, 2017-03-13:
Hi @ninoosewoudt 
First, on with which branch do you encounter a problem (master or testing) ?

ninoosewoudt, 2017-03-13:
both master and testing give the same error.

N0vaPixel, 2017-03-13:
This is strange, i don't have all the informations and knowledge to help you now.
load404 is defined lower
```
void load404() {
  server.send ( 404, "text/html", data_get404());
}
```

A fresh download is working well on my computer, i'm testing on an other one.

ninoosewoudt, 2017-03-13:
i think i fixed it will edit in a bit.

i put the setup function at the bottem so all prototypes would have been called before.

N0vaPixel, 2017-03-13:
This is a fix, you'll never be annoyed. 👍 
But it should work without.
I know in c/c++, all functions must be declared before to be used. But the arduino environement is able to automatically create prototypes of functions.
Seems to be caused by your environement configuration.

tobozo, 2017-03-13:
@ninoosewoudt if you're compiling on linux, move the loop() and setup() at the end of the ino file. It looks like the function hoisting isn't done with this compiler.

I need to do that *every *single *time I update the code from the repository, this does not happen on Windows.

spacehuhn, 2017-03-13:
Oh I didn't know about this bug. You could have made a PR! I'll fix that as soon as I come home - or @N0vaPixel does it if he has time (?) :D

N0vaPixel, 2017-03-13:
No problem 👍  4613963e3c24bd4b4799dee9e6c653d06f7fe5af
Please @tobozo @ninoosewoudt  can you confirm that is now fixed ?

tobozo, 2017-03-13:
@N0vaPixel that's more of a wontfix, this is more related to the toolchain than the project code.
By moving loop() and setup() at the end, it's on OSX and Windows' that the Arduino IDE will complain.

ninoosewoudt, 2017-03-13:
@tobozo i compile on osx and got the error.
@N0vaPixel  i will check if it wont complain anymore.

N0vaPixel, 2017-03-13:
The other solution is to manually add the declaration of all functions at the top and all the definitions at the bottom. But we have too many function.
And i don't see why Windows and OSX ide will complain. Put declaration before loop and definition after or put loop after definition will produce the same result.

spacehuhn, 2017-03-13:
Hm... I actually started this project on a Mac and had no issues compiling 😕

tobozo, 2017-03-13:
You had issues compiling my first PR where I had moved loop() and setup() to the end, same code could not compile on my Windows machine but compiled fine on Linux, so I assumed Windows and Mac had something in common towards, or maybe it's the version number of the IDE?

spacehuhn, 2017-04-03:
This should be fixed now. Can you confirm this?

closed 2017-04-23 by spacehuhn