SpacehuhnTech/esp8266_deauther #119: authorize?

DevShekhawat7, 2017-03-13:
Sorry if this sound dumb and maybe impossible, but can we add the option to to authorize ourselves!
like we can access the internet. Instead of kicking the people from network why not join it, sounds illegal but must be fun to have it :p 
again, sorry if this is dumb question.

H1ghTech, 2017-03-13:
Read the readme: Wpa handshakes can't be sniffed
U can't do this with a 20ct chip! You can with a raspi

spacehuhn, 2017-03-13:
You should learn how Wi-Fi and the different attacks work!
There is no such thing as a master key which can be send to authorize yourself into every WiFi network. There are hacks and vulnerabilities to get access, but these aren't possible with this chip.

closed 2017-03-13 by spacehuhn