SpacehuhnTech/esp8266_deauther #120: QUESTION - What's That probe option?

FrostyModz, 2017-03-13:
I'm on v1.1 and I am wondering, what's that probe option under attacks menu. So. What is it. How do I use it. And what can it be used for

poslogic, 2017-03-13:
Hi,

It is related to this feature request.

https://github.com/spacehuhn/esp8266_deauther/issues/102

tobozo, 2017-03-13:
It performs the [Evil Twin](http://www.kalitutorials.net/2014/07/evil-twin-tutorial.html) attack (or is it [KARMA](https://insights.sei.cmu.edu/cert/2015/08/instant-karma-might-still-get-you.html)?),

spacehuhn, 2017-03-13:
> It performs the Evil Twin attack (or is it KARMA?),

not really. It just asks for non existing networks from a non existing MAC adress. WiFi trackers search for these packets and will think there are dozents of clients nearby.

closed 2017-03-17 by spacehuhn

scarfasss, 2017-03-26:
Is there a practical way to see this probe request spam? My wifi tracker is not picking up any of it when turned on.

Thanks!

spacehuhn, 2017-03-26:
You could use airodump-ng to or any other advanced scanner which picks up client packets.