SpacehuhnTech/esp8266_deauther #121: [Feature Request] Full APScan Async List

tobozo, 2017-03-13:
Ability to get the full AP Scan has not been reached yet, here are the milestones:

- [ ] maxAPScanResults only affects json response, acts as pagination sizer instead (can be set to 1)
- [ ] Web Server event "ApScanResults.json" accepts "apid" as an argument, returns one AP in json 
- [ ] Web Server event "APScanResults.json" returns extra pagination info when ScanResults size is over maxAPScanResults
- [ ] JavaScript Client is able to detect and handle pagination for APScanResults
- [ ] Cancel & Replace (or actually implements) "Continuous Scan" feature

spacehuhn, 2017-03-14:
I think I might found a solution to the problem. We can still send it as one file, instead of cutting it after the maxsize and then asking via JS for every left peace, we can send the full string in splittet parts with the webserver. The browser will recognize it as one file.

I'll show you guys what I mean later. Just got this working on an other project where it's necessary to view large files, which can be bigger than the RAM.

spacehuhn, 2017-03-14:
Ok I guess this isn't easy to implement right away due to the current programm structure, however that's what I did in my other project:

**A few preparations:**
- go to `esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src`   
- edit `ESP8266WebServer.h`
- cut out the `void _prepareHeader` function declaration under `protected:` and paste it under `public:`

**To the actual code:**

I used a buffer array like we have in data.h:
`char data_websiteBuffer[6000];`  
```
void test(){
  int fileSize = 11500;
  int bufferCounter = 0;

  String header; //HTTP header String
  server._prepareHeader(header, 200, (const char*)"text/html", fileSize); //create a HTTP header
  server.sendContent(header);  //send the HTTP header
  
  for(int i=0;i<fileSize;i++){ //fileSize is on purpose bigger than our buffer
    data_websiteBuffer[bufferCounter] = random(256); //make some random bytes!
    bufferCounter++;
    if(bufferCounter >= sizeof(data_websiteBuffer)){ //when the buffer is full then...
      server.sendContent_P(data_websiteBuffer, bufferCounter); //send the current buffer
      bufferCounter = 0; //reset the bufferCounter variable
    }
  }
  
  server.sendContent_P(data_websiteBuffer, bufferCounter); //send the rest of the buffer  
}
```

and then of course:

`server.on("/test", test);`

tobozo, 2017-03-14:
How about using [this](https://github.com/me-no-dev/ESPAsyncWebServer) ?

spacehuhn, 2017-03-14:
Ok I guess that would be way easier and better... You can also load websites directly  out of the progmem!

tobozo, 2017-03-14:
oh this means a complete rewrite of the html/js am I right?
Currently the project is not very friendly with frontend dev iterations, I had to use SerialServer + persistence to get the opportunity to modify the files without reflashing to see my changes and it's soo f*****g slow I'll probably close the Full Serial Peering feature request :-)

But the good thing is a nodejs app would eventually bring the ability to modify html/js/css without reflashing the ESP on every change

It would also enable the possibility to regenerate data.h (or recompile & [flash the ESP](https://www.npmjs.com/package/node-esp) but that's not the point).

How do you handle your frontend iterations? Do you also minify/convert/flash manually on every change?

spacehuhn, 2017-03-14:
Yes I do this everytime. But unless it's something with JavaScript, I test it within the browser with its developer options beforehand.

I don't think we should rewrite everything now, that would be soooooo much work!
The only thing I really wanna see fixed is the APScan list, just some workaround to the current problem - it doesn't need to be pretty, just functional.
One way is the async framework you linked, another would be using my webserver 'hack'. I don't know what's easier though.

I don't have much time to code at the moment and I wanna move on to other projects.

spacehuhn, 2017-03-16:
I can't get the async library working with the 2.0.0 SDK 😞

tobozo, 2017-03-16:
The buffering hack solves the problem for big files coming from data.h, but doesn't solve the problem in getAPScanResults(), where the huge JSON is built and crashes.
Back to pagination then?

spacehuhn, 2017-03-16:
I think the hack can solve that problem too if it writes into the same website buffer instead of a temporary string. But I haven't had time to try it out yet.

spacehuhn, 2017-03-16:
I found a way to use my 'hack' without a change in the library files.
My little tuturial:
https://gist.github.com/spacehuhn/6c89594ad0edbdb0aad60541b72b2388

spacehuhn, 2017-03-16:
Fixed :D a71946d09abecee3e98cf63d9dcd8356e9414b11
It actually was because of the string!

I added 3 new functions to data.h:
```
void sendBuffer();
void sendToBuffer(String str);
void sendHeader(int code, String type, size_t _size);
```

`sendHeader` must be called first, then copy every string (or substring, when generating json files) with `sendToBuffer()` and if everything is done call `sendBuffer`.

tobozo, 2017-03-19:
Nice job! I wish I could test it but I just saw my last wemos die in a tiny mushroom cloud of toxic smoke :fire: :mushroom: :cloud:  . The deauther is still running well but it's an older version and I can't update it anymore (CH340 dead I guess).

Looking at [this thread](https://github.com/esp8266/Arduino/issues/1365) when using `Transfer-Encoding: chunked` you don't have to calculate the size before sending. (TL;DR just send a `Content-Length:0` when it's finished). I'm not sure this applies to sdk2.0.0 but this could save you from looping twice in the results.

tobozo, 2017-03-31:
Closing this thread as very long APSCan list is no longer a stability problem, tested on WemosMini D1 and NodeMCU DevKit V3 without a glitch \o/

closed 2017-03-31 by tobozo

spacehuhn, 2017-03-31:
Great! I'm very happy that it works now! 
I saved the linked thread, however I'm busy with other things right now so I couldn't test it yet.