SpacehuhnTech/esp8266_deauther #122: 1.1 Binary file not working

peekpt, 2017-03-14:
I tried to upload with esptool the  non-512k version 1.1. with no problem but I can't get it to work just got the to led blink every one second.
´´´
esptool.py --port /dev/cu.SLAB_USBtoUART -b 921600 write_flash --flash_mode dio  --flash_size 4m 0x0 esp8266_deauther_1.1_1mb.bin
´´´

But I compiled the code and it worked, so it seems to be a problem on binary file.

spacehuhn, 2017-03-17:
> the led to blink every one second.

sounds like it's crashing and rebooting all the time.
Open a serial connection to the ESP8266 and see what it says. Baud rate is 115200.

peekpt, 2017-03-17:
I tried, sometimes I get "invalid parameters". But I figured out that must be something with  esptool.py, if I try to repeat the arduino command it works fine.
```
/Users/paulo/Library/Arduino15/packages/esp8266/tools/esptool/0.4.6/esptool -vv -cd nodemcu -cb 921600 -cp /dev/cu.SLAB_USBtoUART -ca 0x00000 -cf /var/folders/ll/7skr_7_90p3bf2t3vgpqh7740000gn/T/arduino_build_789848/esp8266_deauther.ino.bin
```
this esptool is not the python version, very strange... 

I also tried to flash with esptool.py my own compiled arduino image with no luck , I'm on mac btw so the possibilities are few.

peekpt, 2017-03-17:
Your image Works! here is the code if you want to try on mac
```
cd ~/Library/Arduino15/packages/esp8266/tools/esptool/0.4.6/
./esptool -vv -cd nodemcu -cb 115200 -cp /dev/<SERIALPORT> -ca 0x00000 -cf <PATH TO BIN FILE>
```

closed 2017-03-17 by N0vaPixel