SpacehuhnTech/esp8266_deauther #123: no matching function for call to 'ESP8266WiFiClass::scanNetWorks(bool, bool&)'

mitch619911, 2017-03-15:
```
Arduino: 1.6.7 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

C:\Users\Mitch\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note:   candidate expects 1 argument, 2 provided

Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Mitch\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi 
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\Mitch\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer 
Using library ESP8266mDNS in folder: C:\Users\Mitch\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266mDNS (legacy)
Using library EEPROM at version 1.0 in folder: C:\Users\Mitch\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\EEPROM 
exit status 1
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'
```
 results = WiFi.scanNetworks(false, settings.apScanHidden);
Ideas?

tpmodding, 2017-03-15:
did you replaced the files with the files described in the readme?

mitch619911, 2017-03-15:
Yes, that fixed it but got a new error:

`Arduino: 1.6.7 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

In file included from sketch\Attack.h:7:0,

                 from sketch\Attack.cpp:1:

C:\Users\Mitch\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:451:7: error: expected initializer before '-' token

 int Wi-Fi_register_send_pkt_freedom_cb(freedom_outside_cb_t cb);

       ^

C:\Users\Mitch\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:452:8: error: expected initializer before '-' token

 void Wi-Fi_unregister_send_pkt_freedom_cb(void);

        ^

C:\Users\Mitch\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:453:7: error: expected initializer before '-' token

 int Wi-Fi_send_pkt_freedom(uint8 *buf, int len, bool sys_seq);

       ^

sketch\Attack.cpp: In member function 'bool Attack::send()':

Attack.cpp:102: error: 'wifi_send_pkt_freedom' was not declared in this scope

   if(wifi_send_pkt_freedom(packet, packetSize, 0) == -1){

                                                 ^

exit status 1
'wifi_send_pkt_freedom' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
`

Also on this, I don't have USP8266, grey out.
```
16 Select your ESP8266 board at Tools > Board and the right port at Tools > Port
If no port shows up you may have to reinstall the drivers.
```

Thanks for your help!!

tpmodding, 2017-03-15:
did you read the readme? and followed the steps? did you edit your user_interface.h how it mentioned in the readme?

mitch619911, 2017-03-15:
Redone everything and now I get
```Arduino: 1.6.7 (Windows 10), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck"

--blah stuff--
 
exit status 1
'load404' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.
```

'load404' was not declared in this scope

tpmodding, 2017-03-15:
https://github.com/spacehuhn/esp8266_deauther/issues/117

mitch619911, 2017-03-15:
So I should reinstall?>

tpmodding, 2017-03-15:
reread https://github.com/spacehuhn/esp8266_deauther/issues/117#issuecomment-286108788

mitch619911, 2017-03-15:
> i think i fixed it will edit in a bit.
> 
> i put the setup function at the bottom so all prototypes would have been called before.
This? - No idea???

N0vaPixel, 2017-03-15:
You seems to compile on linux based OS.
**Solution 1** Install the pre-compiled binary files (releases)
**Solution 2** Use the testing branch, we fixed this bug

spacehuhn, 2017-03-17:
Is your problem solved?

mitch619911, 2017-03-17:
Just installed it on another PC, and that fixed it...

closed 2017-03-17 by mitch619911