SpacehuhnTech/esp8266_deauther #124: WiFiEventHandler does not name a type

scargill, 2017-03-15:
Hi

Copied the instructions - before realising I had to overwrite 2 files it compiled fine - over-wrote the two files and now (this is version 2.30 - reasonably current)

error - WiFiEventHandler does not name a type

(gives that error twice..)

spacehuhn, 2017-03-17:
You need to use v.2.0.0 of the SDK and follow every installation step carefully.

N0vaPixel, 2017-03-17:
This is not the expressif SDK
[https://github.com/esp8266/Arduino](url)

Follow the readme.md file

rajatpunkstaa, 2017-03-29:
I am using official Arduino IDE and esp version 2.3.0 and getting the same error:
Arduino: 1.8.1 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

In file included from xxx/yyy/Downloads/esp8266_deauther-master/esp8266_deauther/esp8266_deauther.ino:5:0:
/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.h:117:3: error: 'WiFiEventHandler' does not name a type
   WiFiEventHandler _disconnectedHandler;
   ^
/xxx/yyy/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.h:118:3: error: 'WiFiEventHandler' does not name a type
   WiFiEventHandler _gotIPHandler;
   ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

Any suggestions ??

Fireflaker, 2017-03-29:
try upload ver1.1 bin file instead of ver1.0 arduino file

Fireflaker, 2017-03-29:
By  the way, i have not successfully flash this board at 115200 baud EVER!
It prints "flash at 9600 baud” on the other side of board. Try so and you
might be amazed.
NodeMCU 1.0,9600,80MHz,4M(does not matter), works for me and is relatively
stable.

Fireflaker, 2017-03-29:
Choose the right compiler.
The compiler is not the default compiler for arduino uno 
I am not exactly sure but it should be called Arduino IDE

closed 2017-04-23 by spacehuhn