SpacehuhnTech/esp8266_deauther #132: Scanning for access points on mobile doesn't work.

ElliotPoots, 2017-03-18:
Hello. I just want to say, I am a huge fan, great work!

I came to use this on my mobile phone (iPhone 5) the other day, and went to scan for my access point, it stated that it was 'scanning' for about 10-15 minutes before I gave up. I don't think it wants to work on the mobile device. I have tested this on my laptop, and the scan is completed successfully in about 5 seconds. I have tried a _wide_ range of different browsers, such as Safari, Chrome, Google (App), Tor, Firefox, Aloha, Opera. None of them work, unfortunately.

Any help would be appreciated. Many thanks.

Burockk, 2017-03-19:
I have the same problem cant use it with iphone phones can you please fix it

spacehuhn, 2017-03-19:
I can't help without knowing what the error is. You have do use a debugger so see what the error is. It's probably some JavaScript incompatibility.
It's working fine one IOS 10.

icsom, 2017-03-20:
I can confirm that it works without problems on IOS 10.2.1

JuncoJet, 2017-03-21:
```
//data.h replace the data_functionsJS[]
const static uint8_t data_functionsJS[] PROGMEM = {
  0x66, 0x75, 0x6E, 0x63, 0x74, 0x69, 0x6F, 0x6E, 0x20, 0x67, 0x65, 0x74, 0x52, 0x65, 0x73, 0x70, 
  0x6F, 0x6E, 0x73, 0x65, 0x28, 0x61, 0x64, 0x72, 0x2C, 0x63, 0x61, 0x6C, 0x6C, 0x62, 0x61, 0x63, 
  0x6B, 0x2C, 0x74, 0x69, 0x6D, 0x65, 0x6F, 0x75, 0x74, 0x43, 0x61, 0x6C, 0x6C, 0x62, 0x61, 0x63, 
  0x6B, 0x2C, 0x74, 0x69, 0x6D, 0x65, 0x6F, 0x75, 0x74, 0x29, 0x7B, 0x69, 0x66, 0x28, 0x74, 0x69, 
  0x6D, 0x65, 0x6F, 0x75, 0x74, 0x43, 0x61, 0x6C, 0x6C, 0x62, 0x61, 0x63, 0x6B, 0x3D, 0x3D, 0x3D, 
  0x75, 0x6E, 0x64, 0x65, 0x66, 0x69, 0x6E, 0x65, 0x64, 0x29, 0x7B, 0x74, 0x69, 0x6D, 0x65, 0x6F, 
  0x75, 0x74, 0x43, 0x61, 0x6C, 0x6C, 0x62, 0x61, 0x63, 0x6B, 0x3D, 0x66, 0x75, 0x6E, 0x63, 0x74, 
  0x69, 0x6F, 0x6E, 0x28, 0x29, 0x7B, 0x6C, 0x6F, 0x63, 0x61, 0x74, 0x69, 0x6F, 0x6E, 0x2E, 0x72, 
  0x65, 0x6C, 0x6F, 0x61, 0x64, 0x28, 0x29, 0x7D, 0x7D, 0x69, 0x66, 0x28, 0x74, 0x69, 0x6D, 0x65, 
  0x6F, 0x75, 0x74, 0x3D, 0x3D, 0x3D, 0x75, 0x6E, 0x64, 0x65, 0x66, 0x69, 0x6E, 0x65, 0x64, 0x29, 
  0x7B, 0x74, 0x69, 0x6D, 0x65, 0x6F, 0x75, 0x74, 0x3D, 0x38, 0x30, 0x30, 0x30, 0x7D, 0x76, 0x61, 
  0x72, 0x20, 0x78, 0x6D, 0x6C, 0x68, 0x74, 0x74, 0x70, 0x3D, 0x6E, 0x65, 0x77, 0x20, 0x58, 0x4D, 
  0x4C, 0x48, 0x74, 0x74, 0x70, 0x52, 0x65, 0x71, 0x75, 0x65, 0x73, 0x74, 0x28, 0x29, 0x3B, 0x78, 
  0x6D, 0x6C, 0x68, 0x74, 0x74, 0x70, 0x2E, 0x6F, 0x6E, 0x72, 0x65, 0x61, 0x64, 0x79, 0x73, 0x74, 
  0x61, 0x74, 0x65, 0x63, 0x68, 0x61, 0x6E, 0x67, 0x65, 0x3D, 0x66, 0x75, 0x6E, 0x63, 0x74, 0x69, 
  0x6F, 0x6E, 0x28, 0x29, 0x7B, 0x69, 0x66, 0x28, 0x78, 0x6D, 0x6C, 0x68, 0x74, 0x74, 0x70, 0x2E, 
  0x72, 0x65, 0x61, 0x64, 0x79, 0x53, 0x74, 0x61, 0x74, 0x65, 0x3D, 0x3D, 0x34, 0x29, 0x7B, 0x69, 
  0x66, 0x28, 0x78, 0x6D, 0x6C, 0x68, 0x74, 0x74, 0x70, 0x2E, 0x73, 0x74, 0x61, 0x74, 0x75, 0x73, 
  0x3D, 0x3D, 0x32, 0x30, 0x30, 0x29, 0x7B, 0x63, 0x61, 0x6C, 0x6C, 0x62, 0x61, 0x63, 0x6B, 0x28, 
  0x78, 0x6D, 0x6C, 0x68, 0x74, 0x74, 0x70, 0x2E, 0x72, 0x65, 0x73, 0x70, 0x6F, 0x6E, 0x73, 0x65, 
  0x54, 0x65, 0x78, 0x74, 0x29, 0x7D, 0x65, 0x6C, 0x73, 0x65, 0x7B, 0x74, 0x69, 0x6D, 0x65, 0x6F, 
  0x75, 0x74, 0x43, 0x61, 0x6C, 0x6C, 0x62, 0x61, 0x63, 0x6B, 0x28, 0x29, 0x7D, 0x7D, 0x7D, 0x3B, 
  0x78, 0x6D, 0x6C, 0x68, 0x74, 0x74, 0x70, 0x2E, 0x6F, 0x70, 0x65, 0x6E, 0x28, 0x22, 0x47, 0x45, 
  0x54, 0x22, 0x2C, 0x61, 0x64, 0x72, 0x2C, 0x74, 0x72, 0x75, 0x65, 0x29, 0x3B, 0x78, 0x6D, 0x6C, 
  0x68, 0x74, 0x74, 0x70, 0x2E, 0x73, 0x65, 0x6E, 0x64, 0x28, 0x29, 0x3B, 0x78, 0x6D, 0x6C, 0x68, 
  0x74, 0x74, 0x70, 0x2E, 0x74, 0x69, 0x6D, 0x65, 0x6F, 0x75, 0x74, 0x3D, 0x74, 0x69, 0x6D, 0x65, 
  0x6F, 0x75, 0x74, 0x3B, 0x78, 0x6D, 0x6C, 0x68, 0x74, 0x74, 0x70, 0x2E, 0x6F, 0x6E, 0x74, 0x69, 
  0x6D, 0x65, 0x6F, 0x75, 0x74, 0x3D, 0x74, 0x69, 0x6D, 0x65, 0x6F, 0x75, 0x74, 0x43, 0x61, 0x6C, 
  0x6C, 0x62, 0x61, 0x63, 0x6B, 0x7D, 0x3B
};
```

ChaudhryAbdulAziz47, 2017-03-27:
A easy fix is to just Download Google Chrome browser on your iphone or any other iOS device and use deauther form there... I've using it for past 2 weeks with no problems! :+1:

ElliotPoots, 2017-03-27:
@ChaudhryAbdulAziz47
Like I stated in my question... I have already attempted Google Chrome.
I'm running on an old iOS build, I presume updating will fix this issue.

spacehuhn, 2017-03-31:
Can you test if the new realease works for you?
https://github.com/spacehuhn/esp8266_deauther/releases/tag/v1.2

Burockk, 2017-03-31:
Tested and it Works now ty so much :)    👍

leesayao, 2017-04-05:
v1.2 fixes the AP scan on iOS with Chrome and Safari, but it seems like deauth attack isn't working. Status shows started at 20pkt/sec but clients remain connected. Beacon clone seems to work. 

Google Chrome	57.0.2987.137 (Official Build) stable (64-bit)
Revision	f4da137433e4...
OS	iOS 
User Agent	Mozilla/5.0 (iPhone; CPU iPhone OS 9_2_1 like Mac OS X) AppleWebKit/601.1 (KHTML, like Gecko) CriOS/57.0.2987.137 Mobile/13D15 Safari/601.1.46

leesayao, 2017-04-05:
Just tried Chrome for Windows 7 also, same thing. I'd love to test any suggestions you may have!

leesayao, 2017-04-10:
Commenting to confirm testing v1.2 fixes the scanning issue on iOS with Chrome. My deauth problem was user error. I have an issue with packet rate settings, however I'll post in the correct issue thread.

scibur314r8t, 2017-04-25:
I hope this in some way helps i tested the latest at time of  build 1.3 (4m) with both android and ios devices the first been a samsung galaxy s3 international. Running android 4.3 open optus Australia firmware. And using chrome as a browser  And there were no issues, it does how ever drop the wifi wene scanning for ap's and i had to wait a short while for it to come back on line or open my wifi settings for it to do so then refresh the page. this was also the same for the nexus 6 international running android 5.1 lolipop network unlocked and the one plus 3 running latest as of 24th/4/17 oxy os.and kali nethunter The ios device used (full disclosure not stock and uses ziperiums ios-t2.5 build ) a iphone 6c AUS/NZ model model produced the same results though did take longer.

N0vaPixel, 2017-08-08:
Since the reopening of the issue section, there has been many changes and bugfix. So i think some issues are outdated now. It better to keep a clean issue section.

This issue seems to have been fixed by dcdec41066dddc2eb64183c57cdf2798c5a2946f

closed 2017-08-08 by N0vaPixel