SpacehuhnTech/esp8266_deauther #133: Disconnects upon deauth

Clon3d, 2017-03-19:
Please read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) 
and the [readme](https://github.com/spacehuhn/esp8266_deauther/blob/master/README.md)
before posting a new issue. Thanks!

Hello, using version 1.1 bin.   whenever I try to death it says starting then loses all connection causing me to have to do a full reboot and does not work against the wifi networks.  I tried version 2.0 but will not compile correctly following your exact instructions. please reply or PM me for details. thanks

Giuseppe2018, 2017-03-21:
i have the same issue...

spacehuhn, 2017-03-22:
What does it say in the serial monitor?

>  I tried version 2.0 but will not compile correctly following your exact instructions.

You need the SDK version 2.0.0. If you get an error than that's an other issue.

RMOA789, 2017-03-23:
Same issue. For some APs it's working well without disconnecting, but for other APs, I get an unavailable webpage. Already tried with different monitors/smartphones.

Flashing with the bin file from the git.

elliot12, 2017-03-31:
Also the same issue. I can deauth an entire network but when scanning for devices on the network to deauth individually the esp8266 restarts. Other than that though, its working 100%. Thanks for the great work @spacehuhn 👍

jLynx, 2017-03-31:
I also have this issue

jLynx, 2017-03-31:
I have noticed, if im connected to the deauther on my computer it works fine, but if I try it on my phone it does exactly what the problem describes. But im not sure, it could just be a coincidence

spacehuhn, 2017-03-31:
> but when scanning for devices on the network to deauth individually the esp8266 restarts.

Your problem is that it's getting disconnected when scanning for clients?

elliot12, 2017-03-31:
Correct. It won't let me see individual devices on a given AP. It just
disconnects and comes on again about 10-15 seconds after but still doesn't
show me the clients. I used the bin file btw.

spacehuhn, 2017-03-31:
It's normal that it disconnects, it has to do that in order to scan for devices.
To find devices they have to send packets while you are scanning, so make sure you generate some traffic in this network (by loading a site for example).
Also note that if your network has 2.4 and 5Ghz it will only find the devices which are using the 2.4Ghz connection.

Fireflaker, 2017-03-31:
Would it be possible to develop it based on ESP32? It is just as cheap and
has 2x 160MHz cores with both bluetooth and wifi. The dual core should make
coding a lot easier.

spacehuhn, 2017-03-31:
The ESP32 isn't able to inject packets, so no :(

jLynx, 2017-04-02:
This is the problem I am having https://imgur.com/iSISaNa

It gets to starting then times out the page when it try to reload. But it does stay connected to the device, as per the WiFi icon in the notification bar. But I cannot access any page from it now until I reset the power

spacehuhn, 2017-04-02:
Yeah... I still need the serial log from you guys.

jLynx, 2017-04-02:
Oops sorry didn't see that message 

ok..typical... it works this time. Anyway here is my log https://pastebin.com/qWd5MFZw

spacehuhn, 2017-04-02:
thank you. The device isn't crashing and doesn't throw any other error.

jLynx, 2017-04-02:
I have tested it 2 other times since that post and both have timed out, no error messages or anything out of the ordinary.

spacehuhn, 2017-04-02:
Sorry I didn't want to close it! (stupid phones..)
Anyway...
can you access the website with another device?

jLynx, 2017-04-02:
Yes on my desktop I have no problems. Just tried again and and it worked perfectly like it should and instantly started deauthing, where before in that log I sent it took at least 30 seconds or more before it started working (Website showing up anyway)

spacehuhn, 2017-04-02:
OK yes that's really weird. It's not like that the ESP8266 doesn't send the website or something, because then it also wouldn't work with your PC.

jLynx, 2017-04-02:
Could it be a way Android's Chrome handles the requests or something?

jLynx, 2017-04-02:
Ok it just happened again while logging, and there is no crashes just the constant running deauth attack message. no matter what I did on my phone I could not access it. I tried disconnecting and reconnecting and that didn't work, but on my desktop I connected to the point and I can see the attack still running and then stop it with no problems.

Now i'm getting timeout (took too long to respond) with a 504 error on my phone trying to load the site, but no errors in the logs in console

spacehuhn, 2017-04-08:
I maybe found a solution.
Scan for the network you want to deauth. Look what channel it uses and then go to settings and set the channel to the same value.
Then restart your ESP8266 and try to deauth that network.

I shouldn't loose the connection then.

jLynx, 2017-04-10:
That seems to fix the problem! Thanks @spacehuhn 

Now do you think it's possible to fix this in software or is it just a problem we will have to live with?

jLynx, 2017-04-17:
@spacehuhn can this issue be closed now?

closed 2017-08-09 by spacehuhn