SpacehuhnTech/esp8266_deauther #138: Changing SSID and password

zak1975, 2017-03-23:
Hello,
I am using Version 1.1.bin. Everything was ok since I tried to change the ssid from pwned to another name. In the same step I also changed the wifi Password and clicked "save".
After that the new ssid is displayed, but when i try to connect, the new password is not accepted. I was very carefully when changing the password and i am sure that i wrote it correctly.
Also the old password "deauther" doesnt work.
I flashed the esp8266 again, hoping that would set everything to default, but after that it Shows the new given ssid again and no Login possible.
Any ideas?

poslogic, 2017-03-23:
Erase your ESP first then flash again, Google "esp8266 erase eeprom"

zak1975, 2017-03-23:
This worked, thank you.
I erased it with esptool.py and then flashed again. Now it is default ssid again.

poslogic, 2017-03-23:
Glad to hear that you got it solved.

Please close this issue 👍

zak1975, 2017-03-23:
I tried again to change ssid and password. This time it worked properly, so the problem is not reproducable for me.
Issue closed.

closed 2017-03-23 by zak1975

cfa999, 2018-12-31:
How do you change the SSID and the password?

developerfromjokela, 2018-12-31:
You should go to settings and change SSID and Password settings.

saintofinternet, 2019-01-02:
mostly it is the timeout error that bugs the process. 

anyways i can change the timeout settings when changing the password or ssid etc?

AN0NIT, 2021-01-04:
> How do you change the SSID and the password?

esp8266_deauther/A_config.h

Chesterphillip, 2021-01-04:
Can I just purchase one complete