SpacehuhnTech/esp8266_deauther #141: General Client Scan Issue

BusesCanFly, 2017-03-25:
First off, phenomenal work! I'm blown away by this!

Everything seems to work flawlessly, except with the client scan. It seems that the scan is a race against how fast your browser decides to error out of the page because you have no internet connection (As the esp has only one antenna). The maximum scan length I can do before chrome decides to error out the page because of no internet is 5 seconds, which does not detect a large amount of clients. Am I making a noob mistake, or is this just something that'll have to be like this?

epiller, 2017-03-25:
The ESP8266 AP is turned off while scanning for clients - which means that your phone/computer will disconnect from it in about 5 seconds. All you need to do is connect to the ESP8266 AP again and refresh the page.

BusesCanFly, 2017-03-25:
Thank you for the response!
Right, this is the client v monitor mode idea, with only on interface. However, after the automatic reconnect, I do not have any clients picked up, even when connected to 15+ AP's. However, in the 5 second window, I can pick up one client sometimes. I doubt its the effect of the browser directly, as its all on the ESP, but maybe the connection registering is resetting the client scan?

epiller, 2017-03-25:
Well, this concludes that I'm not the only one having that issue. I just posted it a few minutes ago, check it out [here](https://github.com/spacehuhn/esp8266_deauther/issues/143).

closed 2017-03-25 by spacehuhn