SpacehuhnTech/esp8266_deauther #143: Client Scan Issue

epiller, 2017-03-25:
## **What's the problem:** 
Whenever I select any WiFi AP and scan for client devices, no matter how much time I set in the "Scan time", the list will almost always be empty.
There were a few exceptions where one or two devices would show up, but a lot (over 8) of them were missing. This is the same for every network I've tried it on.

### **What I'm Using:** 
- NodeMCU v2 (Amica) ESP8266 Board
- esp8266 2.0.0 in the boards manager, so that shouldn't be an issue. 
- Applied the "sdk_fix" and edited the "user_interface.h".

### **What have I tried to troubleshoot the problem:** 
- Engaged all devices inside the network to send/receive information and set the scan time to 100 seconds, so all devices were sending packets in those 100 seconds.
- Tried flashing esp8266_deauther on 3 other (fresh) boards (Same NodeMCU v2 boards) with no success.
- Tried using older builds of esp8266_deauther
- Tried using different power supplies
- Tried switching browsers, devices and operating systems
- Failed

### Notes:
Not-such-a-big-issue: If there's a network selected, I can't access settings. The page will load forever and eventually quit, but if I click back on "APs", "Clients" or "Attack", it loads them perfectly fine.
Also, mad respect to spacehuhn and contributors for developing this, can't imagine the time that went into it. Thank you for this, honestly.

BusesCanFly, 2017-03-25:
I am having the same issue, as we discussed a few hours ago, (issue [#141](https://github.com/spacehuhn/esp8266_deauther/issues/141)), and I am on a node v3 (ebay seller sent the wrong board...but hey, who's complaining?)

epiller, 2017-03-25:
I have one "Lolin NodeMCU v3 new" fake board too, and I've just tried deauther on it. Got exactly the same issue as mentioned above, so I guess it isn't board related.

spacehuhn, 2017-03-25:
Hello you two!
No it's shouldn't be board related.

Thank you for the details, most people don't do that!

The clients can only be registered if they're active, sending in 2.4Ghz and their signals are in range.
So just keep that in mind!

What you can test is looking at the serial monitor. Maybe the output can tell us more.

epiller, 2017-03-25:
Hey there,

I've noticed that, so I tried to make it easier for you :D
And yep, I know! The router only supports 2.4GHz anyways, and all clients were actively transferring files, all 10 of them (esp, clients and the router were all in a 3m radius).

I'll check the serial monitor in a few minutes and update.

Thanks!

epiller, 2017-03-25:
Got the serial output! Updated the last comment.

spacehuhn, 2017-03-25:
have you selected multiple networks? It switches the channels..

epiller, 2017-03-25:
Nope! Only one. Just checked again.
But even when I select multiple networks, I still get no results (or maybe 1 or 2 if I select 15+ networks).

spacehuhn, 2017-03-25:
No I mean in the log it says that it's switching channels. If you only select one network, it shouldn't do that.
Maybe that's the bug.

epiller, 2017-03-25:
Seems to be the only explaination! 
Is it hard to fix?

spacehuhn, 2017-03-25:
Ok I updated the testing branch. Can you test if this works for you?
https://github.com/spacehuhn/esp8266_deauther/tree/testing

epiller, 2017-03-25:
Just tested it, and it works! Seems to be fixed for the most part now! I can see almost all devices when I select my network, but when I select multiple networks (10 networks for 90 secs), it shows only a few (3) clients, but there should be way more, especially considering that my network is selected in those 10 networks, so it should be way more then 3. 
I'm not sure how to specify the problem now, but here's the serial output: 

> starting client scan
set channel to 1
changing to channel 4
changing to channel 6
changing to channel 9
changing to channel 10
changing to channel 11
stopping client scan after 90000ms
0: 539 Device 1 xx:xx:xx:xx:xx:xx
1: 32 Device 2 xx:xx:xx:xx:xx:xx
2: 3 Device 3 xx:xx:xx:xx:xx:xx
starting WiFi AP
SSID: my_ssid
Password: my_password
getting client scan result JSON 
{ "clients":[{"i":0,"p":539,"m":"xx:xx:xx:xx:xx:xx","n":"","v":"Device 1","s":0,"a":"984d92"},
{"i":1,"p":32,"m":"xx:xx:xx:xx:xx:xx","n":"","v":"Device 2","s":0,"a":"984d92"},
{"i":2,"p":3,"m":"xx:xx:xx:xx:xx:xx","n":"","v":"Device 3","s":0,"a":"984d92"}] }
done

spacehuhn, 2017-03-25:
Well 10 networks in 90 seconds means only 9 seconds per network, which means less time to actually find clients per network.
This client scan is not build for that. It was build to scan one single network and I added the abillity to scan multiple networks.

epiller, 2017-03-25:
Oh, sorry, I thought it was threaded or something, but now I understand. 
Everything works fine then, so thank you. You're awesome.

closed 2017-03-25 by epiller

spacehuhn, 2017-03-25:
No :( but thank you very much for your help! Another bug found and fixed 👍