SpacehuhnTech/esp8266_deauther #147: Problem on program device with arduino ide 1.8.1 1nd esp 2.0

bolkan, 2017-03-26:
Hi,
I use arduino ide 1.8.1 on osx 10.12.3 .
Normally I use esp 2.3.0.
If I downgrade to 2.0 the compilation of the sketch is ok but the program of the module (nodemcu v1.0) fail with this message:

esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.wchusbserial143120
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/pt/mh0hkt3n4s56spj8vt26vrjw0000gn/T/arduino_build_325719/esp8266_deauther.ino.bin success
opening port /dev/cu.wchusbserial143120 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x01 0x07, expected 0x01 0x08
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
warning: espcomm_send_command: wrong direction/command: 0x01 0x07, expected 0x01 0x08
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed

spacehuhn, 2017-03-26:
Have you already tried different settings?

bolkan, 2017-03-26:
Yes wemos d1 -- pure esp8266 12e , but nothing. If can help , for another project I  have put the beta 2.3.x and I have get the same problem. But your project is more interesting and I need to solve my problem. I hope anyone can help me. Best regards.

spacehuhn, 2017-03-26:
You could flash the bin file directly, as described in the readme.

bolkan, 2017-03-26:
With direct flash, after start esp give this via serial 

�AacCcƱ!q�V�a�qae11AaAaa�svDaf�!�a�a�gc!aa��!us1��!e�!��1A
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
system param error
66 70 6d 20 77 6b 20 65 72 72 3a 20 66 70 6d 20 6e 6f 74 20 6f 70 65 6e 0a 00 00 00 00 00 00 00 
66 70 6d 20 77 6b 20 65 72 72 3a 20 70 6d 20 69 73 20 6f 70 65 6e 0a 00 00 00 00 00 00 00 00 00 
77 6b 20 66 72 6f 6d 20 75 73 65 72 0a 00 00 00 71 75 69 74 53 6c 70 50 6f 72 63 0a 00 00 00 00 
71 75 69 74 53 6c 70 50 6f 72 63 0a 00 00 00 00 6c 69 73 74 5f 6e 75 6c 6c 0a 00 00 00 00 00 00 
73 25 64 0a 00 00 00 00 00 00 00 00 00 00 00 00 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
4c 6d 61 63 52 78 42 6c 6b 3a 25 64 0a 00 00 00 4c 6d 61 63 54 78 42 6c 6b 3a 25 64 0a 00 00 00 
79 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 6d 61 70 20 66 61 69 6c 0a 00 00 00 00 00 00 00 
63 68 67 5f 42 3a 25 64 0a 00 00 00 00 00 00 00 10 00 00 00 00 00 00 00 3d 3d 64 6c 0a 00 00 00 
61 64 64 20 6d 69 73 6d 61 74 63 68 0a 00 00 00 61 64 64 20 25 64 0a 00 72 6d 20 25 64 0a 00 00 
72 6d 20 6d 69 73 0a 00 00 00 00 00 00 00 00 00 65 72 72 6f 72 3a 20 70 6c 6c 5f 63 61 6c 20 65 
78 63 65 65 64 73 20 32 6d 73 21 21 21 0a 00 00 25 64 2c 20 00 00 00 00 25 64 2c 20 00 00 00 00 
25 64 2c 20 25 64 2c 20 00 00 00 00 00 00 00 00 25 64 2c 20 25 64 2c 20 00 00 00 00 00 00 00 00 
25 64 2c 20 25 64 2c 20 00 00 00 00 00 00 00 00 25 64 2c 20 25 64 2c 20 00 00 00 00 00 00 00 00 
25 64 2c 20 00 00 00 00 00 00 00 00 00 00 00 00 72 78 69 71 5f 72 65 67 3a 20 69 3d 25 64 2c 20 
61 5f 6d 69 73 3d 25 64 2c 20 70 5f 6d 69 73 3d 25 64 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 
74 6f 74 61 6c 5f 70 77 72 3d 25 6c 64 2c 20 6d 61 78 3d 25 6c 64 2c 20 0a 00 00 00 00 00 00 00 
25 64 5f 25 64 5f 25 64 0a 00 00 00 00 00 00 00 73 65 74 5f 72 78 5f 67 61 69 6e 3a 20 72 66 74 
78 3d 25 78 2c 20 72 66 72 78 3d 78 25 78 2c 20 61 74 74 3d 25 64 2c 20 74 78 62 62 3d 30 78 25 
78 2c 20 62 62 72 78 31 3d 30 78 25 78 2c 20 62 62 72 78 32 3d 30 78 25 78 2c 20 74 64 63 3a 25 
64 2c 25 64 0a 00 00 00 00 00 00 00 00 00 00 00 69 6e 64 65 78 3a 20 25 64 20 76 61 6c 75 65 3a 
20 30 78 25 78 20 30 78 25 78 20 30 78 25 78 20 30 78 25 78 0a 00 00 00 00 00 00 00 00 00 00 00 
0a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 42 4b 3a 25 64 2c 25 64 2c 25 64 2c 25 64 2c 30 
78 25 78 0a 00 00 00 00 00 00 00 00 00 00 00 00 74 78 64 63 3a 25 64 2c 25 64 3b 20 72 78 64 63 
3a 25 64 2c 25 64 2c 25 64 2c 25 64 0a 00 00 00 64 6f 6e 27 74 20 75 73 65 20 72 74 63 20 6d 65 
6d 20 64 61 74 61 0a 00 00 00 00 00 00 00 00 00 25 33 64 2c 25 33 64 3b 20 00 00 00 00 00 00 00 
74 78 69 71 5f 67 61 69 6e 3d 25 64 2c 20 74 78 69 71 5f 70 68 61 73 65 3d 25 64 0a 00 00 00 00 
54 58 50 57 31 2c 20 25 64 2c 20 25 64 2c 20 25 64 2c 20 25 64 2c 20 25 64 2c 20 25 64 0a 00 00 
54 58 5f 50 57 43 54 52 4c 2c 20 74 6f 74 5f 70 3d 25 64 2c 20 64 6f 63 
00 47 32 35 27 34 b7 0c 03 22 a0 00 0d f0 00 00 40 55 11 50 5c 41 67 03 b1 1b 88 06 eb ff 00 00 
f6 da e8 00 0a 40 00 b4 a1 40 95 81 50 40 91 0c 05 16 bb fa a0 99 20 46 e9 ff 67 03 52 60 33 20 
50 33 11 30 3b 41 70 a8 c0 f6 ca 53 00 0a 40 92 a0 00 90 92 81 20 23 81 30 30 91 50 b6 11 b0 55 
30 20 24 c0 30 35 c0 27 b4 01 0b 33 90 90 60 8c 59 0b 53 20 35 83 0b 22 30 a4 a5 87 9a 49 d6 a9 
00 1b 22 16 52 07 f0 99 11 16 79 06 0d f0 00 00 40 33 11 30 3c 41 67 05 ac 1b 77 c6 e9 ff 00 00 
f6 da 18 00 0a 40 00 b2 a1 20 93 81 30 20 91 50 36 11 16 bb fa a0 99 20 46 e9 ff 00 50 76 11 70 
35 30 2d 04 0d f0 00 00 40 83 11 bc 18 60 f8 40 80 8c 41 a7 a6 4e 1b 66 00 16 40 20 88 81 90 22 
81 00 99 a1 30 34 b5 60 33 c0 c0 33 01 8a 33 c6 e2 ff 00 00 20 21 41 f0 22 11 0d f0 1b 33 0d f0 
60 f2 40 62 c6 15 67 2a 1b b6 c6 cb 00 16 40 90 82 81 00 29 a1 0c 09 30 34 b5 60 33 c0 c0 33 01 
8a 33 06 d6 ff 6d 0a 86 f7 ff 00 00 f0 33 11 30 31 41 20 a3 20 16 4a 08 9c 53 a0 f3 40 a2 ca f5 
00 1a 40 20 33 81 00 22 a1 0c 18 a0 88 c0 86 36 00 a0 f2 40 a2 ca f5 6c 18 a0 88 c0 00 1a 40 96 
9a 00 00 32 a1 0c 02 46 30 00 00 00 20 30 91 00 22 a1 86 2d 00 f0 55 11 50 51 41 40 a5 20 bc ba 
9c 55 a0 f5 40 a2 ca f5 00 1a 40 40 55 81 00 44 a1 0c 19 a0 99 c0 46 25 00 a0 f4 40 a2 ca f5 6c 
19 a0 99 c0 00 1a 40 96 9a 00 00 54 a1 0c 04 06 1f 00 00 00 40 50 91 00 44 a1 46 1c 00 70 3f 05 
10 33 01 0c 02 46 64 00 cc d4 f0 85 11 cc 88 41 a1 fe 40 33 20 46 60 00 00 67 c5 0b 40 85 11 40 
88 20 8c 28 3d 05 2d 04 70 7f 05 f0 33 11 00 41 40 30 37 81 86 58 00 00 56 82 fe f0 83 11 56 28 
fe 71 94 fe 70 35 20 c6 53 00 00 00 12 c1 e0 c9 41 d9 51 e9 61 f9 71 61 8e fe 50 73 30 67 43 a7 
67 45 d4 30 84 a5 50 94 a5 16 f8 ef 16 59 f4 9a 88 a1 89 fe 60 33 20 a0 33 10 60 55 20 a0 55 10 
79 11 20 c0 f5 40 d0 f5 30 e0 f5 50 f0 f5 d0 a2 c1 40 bc c1 0c 09 ba aa b7 ba 01 1b 99 10 40 40 
a0 69 81 40 b2 c1 00 aa a1 ba aa b7 ba 02 62 c6 01 0c 09 50 b2 c1 ba 66 b7 b6 01 1b 99 d0 bc c1 
b0 66 80 b7 b6 01 1b 99 40 b3 c1 ba 66 b7 b6 01 1b 99 8c 3a 0c 1b b0 66 20 f0 72 c1 40 be c1 0c 
0a 7a bb 77 bb 02 a2 ca 01 50 7c c1 70 bb 80 77 bb 02 a2 ca 01 d0 73 c1 70 bb 80 77 bb 02 a2 ca 
01 b0 aa 81 9a aa 00 bb a1 ba 66 b7 b6 02 a2 ca 01 0c 09 f0 2c c1 aa 22 a7 b2 01 1b 99 50 a3 c1 
a0 22 80 a7 b2 02 92 c9 01 d0 ae c1 aa 22 a7 b2 02 92 c9 01 f0 b3 c1 50 7e c1 0c 0a 70 bb 80 77 
bb 02 a2 ca 01 b0 aa 81 9a aa 00 bb a1 ba 22 b7 b2 01 1b aa f0 3e c1 aa 33 78 11 0c b4 30 59 41 
cc 45 42 c4 01 82 c8 ff 00 14 40 20 33 81 60 22 81 00 66 a1 42 a3 ff 40 88 c0 f0 44 11 47 b8 34 
d6 86 00 1b 22 ac 72 f0 66 11 9c 96 c0 88 01 8a 33 70 7f 05 10 77 01 70 33 20 c8 41 d8 51 e8 61 
f8 71 12 c1 20 0d f0 20 21 41 f0 2  ld����Ǆ��$�������s ���l���d`�dl������l��l��| �����l�d$���p ��ddl$��l`
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

bolkan, 2017-03-26:
NOW WORK following the correct flash command for nodemcu (esp8266 12E):
sudo esptool.py --port /dev/cu.wchusbserial143120 --baud 115200 write_flash --flash_mode dio --flash_size 32m 0x0 esp8266_deauther.bin

But remain my original problem .....

Hwzsnow, 2017-04-23:
You need a esp8266 ，hahai ，do you know esp8266 drivea on windows 10？ 
is hc340？I very need help

Fireflaker, 2017-04-25:
什么问题？win10不应该有问题，不需要驱动程序，浏览器访问到地址就好

Hwzsnow, 2017-04-28:
问题已经解决了，不过源码编译的时候还是有些编译不过

发自网易邮箱大师

spacehuhn, 2017-04-28:
@Hwzsnow @Fireflaker english only!

Hwzsnow, 2017-04-28:
where are you from

发自网易邮箱大师
On 04/28/2017 19:06, Stefan Kremser wrote:

@Hwzsnow@Fireflaker english only!

Fireflaker, 2017-04-28:
Sorry for that 抱歉中文
wherever you think im from 您觉得呢
You need to first install ESP8266 board compilers from Board manager in
Arduino IDE 您需要根据首页上教程安装汇编器后才能成功编译

killthekid, 2017-04-30:
I have some issues as well.. i flashed the esp (some different kind of esp with external antenna connection) with the esptool.py, after i flashed the firmeware it works, i get acces to the esp and can scan, attack and so on, but once i disconnect the power supply, it seems that everything is lost. 
any ideas? 
[image]

homeflat, 2017-08-20:
i have problem i m using windows7 and arduino 1.8.3 and when i upload esp8266 deauther to my modemcu8266 i get the following error(attachment) what is the problem and what i can do to solve this problem
[error text.txt](https://github.com/spacehuhn/esp8266_deauther/files/1237417/error.text.txt)

N0vaPixel, 2017-08-21:
@homeflat
This is not the proper place to ask this question.
To answer, the given error show that you didn't follow the tutorial correctly.

closed 2017-09-21 by tobozo

SandhiyaB, 2018-11-01:
Error: 
esptool v0.4.13 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 9600
	setting port from COM1 to COM3
	setting address from 0x00000000 to 0x00000000
	setting erase size to 0x00080000
error: Failed to open COM3
error: espcomm_open failed
error: espcomm_open failed

Program: 
#include <dummy.h>
void setup()
{
  pinMode(14, OUTPUT);

}

void loop()
{
  digitalWrite(14, HIGH);
  delay(4000);
  digitalWrite(14, LOW);
  delay(4000);

}