SpacehuhnTech/esp8266_deauther #150: Request

gremlinflat, 2017-03-27:
Can u add whitelist feature??

spacehuhn, 2017-03-27:
please explain that

ChaudhryAbdulAziz47, 2017-03-27:
I Think @gremlinflat means; adding a feather where we can attack other device on network and whitelist the device we don't want to attack. But what he doesn't know is that, we can scan for clients and only attack the device we want! Technically this feature is already there! :+1:  @spacehuhn

gremlinflat, 2017-03-27:
Deauth entire network, without deauth device on list

gremlinflat, 2017-03-27:
The simplest word is the opposite of blacklist

spacehuhn, 2017-03-27:
That's not possible though. You can either deauth all or just specific clients (in which case you need to scan first).

closed 2017-03-27 by spacehuhn

gremlinflat, 2017-03-27:
Have u ever use mdk3 or aireplay in linux?
Its like blind deauth but didnt attack mac addres that listed on whitelist