SpacehuhnTech/esp8266_deauther #151: Version 1.1 is unstable and does not work after reflashing

Fireflaker, 2017-03-27:
It worked fine for 1 day and suddenly the 192.168.4.1 is not accessible. However, the wifi signal is still there. The on board blue led flashes about every 10 seconds and seems to be resetting itself.

I tried to flash it again with the v1.1 bin file and still does not work.
I flashed v1.0 file using Arduino and it worked fine.

What is the problem？

ErfanDL, 2017-03-28:
format your ESP8266 memory and reprogram

Fireflaker, 2017-03-28:
Fixed it
A hardware problem.
Using Node Mcu board ver1.0 using the sop package programmer.
Must hold down the flash button, press reset botten while flash button held down, flash it with 9600 baud while flash button held down, let go flash botten and never touch any of these battens again. Unpower the board, plug it in again and it should work.

spacehuhn, 2017-03-28:
great :)

closed 2017-03-28 by spacehuhn