SpacehuhnTech/esp8266_deauther #157: I CANNOT FLASH THIS FILE PLUS HOW TO RESET THIS BOARD

Vaibhavland, 2017-03-28:
PLZZ GUYS HELP ME WITH THIS BOARD I CANNOT FLASH ANY FILE AND PLUS I CANNOT BURN THE BOOTLOADER PLZZ GIVE ME A GUIDE I M A POOR BOY I CANNOT AFFORD THIS AGIN LITREALLY AND ITS NOT SHOWING ANY WIFI SIGNALS TOO 
vaibhavs362@gmail.com
---

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

---

poslogic, 2017-03-28:
Stop shouting!!!!!

poslogic, 2017-03-28:
Everything is explained here https://github.com/spacehuhn/esp8266_deauther/blob/master/README.md

If you have problems still please follow these recommendations for asking for help https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md

Vaibhavland, 2017-03-28:
Bro i want an bin file for the compatiable esp8266 that we get in market with golden lines on top of it

Vaibhavland, 2017-03-28:
Bro hwo to reset that too??

poslogic, 2017-03-28:
I have no idea what ESP board that you have?
But if you follow the instructions this should work for any type (it has been tested on most)

To reset your board Google "ESP8266 erase eeprom"

If you are struggling with this information try searching YouTube for some basic videos first before attempting this project.

Good luck

spacehuhn, 2017-03-29:
I blocked him.

spacehuhn, 2017-03-29:
Next time don't even help such a spammer.

closed 2017-03-29 by spacehuhn