SpacehuhnTech/esp8266_deauther #160: AP deauth works but Client deauth not

WiredLife, 2017-03-31:
Hello :)
I've build and flashed the newest version to my wemos d1 mini and it works perfect except the client specific deauth.
Maybe i must set another deauth reason in settings for that?
Thanks!

spacehuhn, 2017-03-31:
So if you deauth as broadcast (without selecting a client) it works?

BioEnergy01, 2017-04-18:
After selecting a client on selected AP and going to the attack tab my device reboots. Deauthing an AP works just fine.

spacehuhn, 2017-04-20:
Add any more information.
 Otherwise I will close it as invalid.

> Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

tufaly, 2017-04-24:
I've been having the same problem it seems because could choose the client but the next day when am searching for clients and I reconnect it seems to have restarted

closed 2017-04-29 by spacehuhn