SpacehuhnTech/esp8266_deauther #161: Settings in v1.1 new SSID and hidden wifi

John13K, 2017-03-31:
Hello team,

I follow the setup tutorial Wifi 'Jammer' Device V1.1 from Seytonic and it works fine at first.
But later I change the settings with a new SSID and hidden the wifi and know I can not see the new or the old SSID of the ESP8266 in any device.
I did several times flash using the nodemcu-flasher in order to solve the issue but unsuccessfully.
It seems that the SSID is hidden and I do not know how to see it again !
Thanks for the support

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

---

poslogic, 2017-04-01:
You will need to set your connecting device to connect to the hidden SSID that you set on the settings page.
Look in your network connections (or Google) for connecting to a hidden SSID

John13K, 2017-04-01:
I understand but I do not remember the hidden SSID and I want to come back to default settings.
My goal is to  change the settings in order to go back default settings i.e. no hidden SSID.
What are the exact steps that I need to do go back default settings ?

Thanks for the support

spacehuhn, 2017-04-01:
search for erase eeprom or clear eeprom

John13K, 2017-04-01:
OK thanks I Google erase/clear eeprom method and to be honest the is too much for me I have no knowledge for that level !
Is there a way to rename i.e. change the hidden SSID ? 
I just need to change the hidden SSID in order to allow me to connect to the hidden SSID.

Thanks team for the support

Mcewen123, 2017-04-02:
In the Arduino go to : 
File
then select : Exemples
then EEPROM
then eeprom_clear 
then upload and you are done
[image]

John13K, 2017-04-02:
OK thanks, I could only install the version v1.0 but I could not install v1.1 or v.1.2.

FYI - as lessons learned all my steps:

1º I installed the v1.0 with Arduino upload file and it works 100%.
2º I installed the v1.1 with flash using the nodemcu-flasher and it works 100%.
3º In the v1.1. I change the settings put hidden SSID and I created the problem because I forgot the SSID.
4º I did several times (for the v1.1 and v.1.2) flash using the nodemcu-flasher but does not work.
5º I did several times also upload file with Arduino (for the v1.1 and v.1.2) but does not work neither.
6º I came back to previous step and I reinstalled the v1.0 with Arduino upload file and it works 100%.

It seems that after the settings change with hidden SSID we can not go back.

Once again thanks all for the support

spacehuhn, 2017-04-02:
Yes because since v1.1 it uses the EEPROM to save the hidden-setting. So clear your EEPROM and it should be fine.

spacehuhn, 2017-04-03:
Look in the serial monitor. The ESP8266 will print out its settings so you can see waht SSID it has.

Fireflaker, 2017-04-03:
I had the same issue with you and solved it.

For me, the data stored was damaged for some reason. I could connect to it but it resets itself every 5 seconds. 192.168.4.1 does not respond. Reason for data loss is likely to be frequent power removal and playing with the in board keys too much.(Wrong gramma)

Solution:
1 use a paper clip to clip down "flash"key. Don't take it off during entire time.
2 plug into computer
3 press "reset" key once. If the blue led onboard flash ONLY once, you should be good. IF the led goes crazy or even stays on, check if flash key if held down.
4 flash it with settings: 9600 baud, 80MHz, 4M. MUST BE 9600 or will not work.
5 DO NOT TOUCH RESET KEY. To reset, unplug from power source, remove paper clip, and plug it in again.
6 It worked for me. Good luck. Don't touch any keys when using! Don't reset using reset key. Don't remove power when anything is going on. Or, it might be all good if you are lucky, but if it is accessing eepram, it is dead and you must flash it again.

John13K, 2017-04-03:
problem solved closed :)
The "flash" did not work for me thanks you anyway. 
So as backup I followed the spacehuhn tip and the serial monitor printed the settings with the SSID and now it works 100%.

FYI - The ESP8266 it's a very good device because during this issue I did "thousand times" flash, resets power on/off and  the ESP8266 still working. 

thanks all for the support

schinfo, 2017-04-19:
My tool can fix the problem too.

https://github.com/schinfo/EEPROM-Cleaner

closed 2017-04-23 by spacehuhn