SpacehuhnTech/esp8266_deauther #165: Add Temparature unit

haxord7, 2017-04-01:
#164 
 
Esp8266 is bult in temparature sensor ,sir please add temparature unit it deauth attack so that when we are going to perform deauth attack it will show what the temparature of mcu .

mcmasterp, 2017-04-01:
I didn't know it had a temp sensor. That will be useful in other projects for sure. What use do you see for it in this project? I'd think it may make the difference of what priority the request deserves

haxord7, 2017-04-01:
https://nurdspace.nl/ESP8266 

go here and read it carefully

haxord7, 2017-04-01:
sometime mcu is going hot when er going to send 100+ pkts  ,another reason attacking -89db signals  wifi mcu going hot we need to see temp then we should realize time to slow down attacking, its needed

mcmasterp, 2017-04-01:
Great link and use case. I hope it gets high priority

spacehuhn, 2017-04-01:
It doesn't has an internal temperature sensor.
https://github.com/esp8266/Arduino/issues/380

closed 2017-04-01 by spacehuhn