SpacehuhnTech/esp8266_deauther #167: Feature Requests

spacehuhn, 2017-04-02:
Ideas:
- Serial interface (https://github.com/spacehuhn/esp8266_deauther/issues/37)
- improved SSID naming (https://github.com/spacehuhn/esp8266_deauther/issues/158 https://github.com/spacehuhn/esp8266_deauther/issues/137)
- logging (https://github.com/spacehuhn/esp8266_deauther/issues/130)
- change MAC adress (https://github.com/spacehuhn/esp8266_deauther/issues/71)
- add clients manually

Will not be added:
- auto deauth all access points in range
- sniff for handshakes (not possible)

Fireflaker, 2017-04-03:
Is it possable to do sniff with ESP32? Probably much harder to program though.
Would it be fun to connect it to a transistor amplifier and run beacon attack + channel  hopping so that almost everything running at 2.4G will be satisfied. By that, it becomes an actuall jammer. (Hopefully)
Powerful amps are expensive. You may use output stage on a router or anything with a UHF transistor amplifier.

spacehuhn, 2017-04-03:
@Fireflaker But that's not what this project is meant for.

Fireflaker, 2017-04-03:
🌚😈🌚

bildos99, 2017-04-05:
Implement: arbitrary code execution on the Broadcom Wi-Fi chip

Source: https://bugs.chromium.org/p/project-zero/issues/detail?id=1046#c2

ali6836, 2017-04-17:
Would it be possible to implement adding password(s) to the wifi beacon clone, like in settings have a setting for wifi beacon clone password?

JEmlay, 2017-04-20:
It would be nice if it could remember the SSID list we create.  Or let us create a list in settings then add CUSTOM along side CLEAR/RANDOM/CLONE/ADD

spacehuhn, 2017-04-20:
@JEmlay
https://twitter.com/spacehuhn/status/841403058378682368

There is a save button at the bottom.

jLynx, 2017-04-20:
@spacehuhn no need to be a dick about it

JEmlay, 2017-04-20:
@spacehuhn - you can sigh all you want but when I power off the unit and turn it back on, no more list.

So maybe find out a little more about the problem instead of firing off the childish memes?

Also, having a list in the settings like I "requested" would allow us to go back and forth between custom list and full random.  From what I can tell, even if it did save properly on my unit if I were to blow it out for a full random list then I would have to type in my custom list all over again.

MAPReiff, 2017-04-23:
@spacehuhn idea - When you do the beacon spam/duplication, can we possibly have any devices that connect to it be taken to an HTML file stored on the device?

raphael613, 2017-04-23:
Hi,
First of all good work.
Two things:
1)How can I erase this dot before each SSID?
2)How can I change that the SSID show up "Not Secure" (No wifi-key to connect)?
(The fact that the SSID shows up secure people will not try to connect they do not know the key, even there is no AP...)

I have no problems to learn the code but before I wanted to know first if it is possible to do that through the code or it is the hardware to change something...

Thanks in advance

battletux, 2017-04-28:
Would it be possible to piggy back 2 nodeMCU's in a master/slave config? Where you connect to the Master's web GUI and it relays all scan requests and attacks over serial to the slave which does all the work?

This way it would bypass the whole ap dropping when scanning for clients and allow the web GUI to be a little more responsive.

tobozo, 2017-04-28:
If you're ready to trade the Wifi web-server part for a Serial web server, see #37

snipercon55, 2017-04-29:
would it be possible to intersept the communication between the router and the device to see what people are searching or using the internet for??

spacehuhn, 2017-08-08:
Since most features are implemented now, I gonna close this issue now.
There isn't much more to add at this point. All important things that are possible with this software and hardware setup were implemented.

closed 2017-08-08 by spacehuhn

itsik-b, 2018-07-05:
on future versions, will the device will be able to connect to a local network and to be controlled locally ? 
that way you can control it and don't lose your connection to the world :)

itsik-b, 2018-07-05:
there is an other version that rely on 1.6 ver of this project. [Wi-PWN](https://github.com/samdenty99/Wi-PWN)
the wifi client implemented there and works fine, even managed to do wireless upgrade to the esp8266.