SpacehuhnTech/esp8266_deauther #169: forever hidden

mmsminiclip, 2017-04-02:
i recently created a hidden ssid which i do not remeber the name. I read a recent read saying that settings are saved. This would make sense because ive tried reflashing and the wireless server never pops up anymore :(. Is there a possible way to reformat the nodemcu v3? i should mention that i was using the 1mb version and when i reflash the 512mb version it works but i cant make any changes to the settings so i dont want to settle for that version.

poslogic, 2017-04-02:
To reset your board Google "ESP8266 erase eeprom"

mmsminiclip, 2017-04-02:
i reset my epprom in the arduino ide. But it still isnt showing up when i reinstall the script

spacehuhn, 2017-04-03:
Look in the serial monitor. The ESP8266 will print out its settings so you can see waht SSID it has.

Fireflaker, 2017-04-03:
you need to hold down flash key when flashing it and use 9600 baud. Everything will be erased. Works for me'

SaltyViper, 2017-04-08:
> you need to hold down flash key when flashing it and use 9600 baud. Everything will be erased. Works for me'

This wont erase the EEPROM where the settings are stored

schinfo, 2017-04-19:
... or you can use my tool ;-)

https://github.com/schinfo/EEPROM-Cleaner

closed 2017-04-23 by spacehuhn