SpacehuhnTech/esp8266_deauther #172: Safari iPhone Issues

avoiceofreason, 2017-04-03:
Deauther 1.2 on 4mb ESP dev board

I have a sense that Safari on iPhone is not as consistent as using deauther with android.
Note this is an upgrade from previous version, so eeprom settings would be preserved

1.Deauther attack by client doesn't appear to work. Gives x pkts/s but no actual deauther of client
2.Settings not displayed in UI. Although assume default settings are in operation.
3.Possibly list of AP's and list of clients after respective scans is not as reliable as use with android browser (this may be more perceived than actual)

iPhone 6 with IOS 10.3

poslogic, 2017-04-03:
I agree with you!

People using this should be aware of that they might need to swap and or change phone OS,s and browsers and be prepared to experiment and help with development.

haxord7, 2017-04-07:
i am using chrome , i didn't face this type of problem

Spektrumz, 2017-04-20:
The settings won't load for me and I used 4 different operating systems.

spacehuhn, 2017-08-09:
I think this problem is solved now.

closed 2017-08-09 by spacehuhn