SpacehuhnTech/esp8266_deauther #174: Crashing when going to settings

jLynx, 2017-04-03:
Here is the log: https://pastebin.com/fZr3bRRq

I'm not sure exactly what is causing this. All I did was deauth 3 of my networks, stopped it, scanned for clients, found none then clicked settings and it crashed. Now its on a constant crash loop whenever my phone tries to join and load the settings page (could happen on other pages, I am not sure)

If you need any more information just ask.

jLynx, 2017-04-05:
Any help with this?

tobozo, 2017-04-05:
If you don't get any help, it's probably because your serial log does not say how many access points you have in your area, if you tried to erase flash + re-upload the sketch, if you used the binary or compiled it yourself, what hardware you used, and many other useful information that could help anyone reading this to provide you with an answer.

Also Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.
,

mcmasterp, 2017-04-05:
Also people are busy and not every issue can be a priority. Definitely no post is missed on most gits so my method is to either add more info to my post as i go or wait wait wait lol. It's at these times i remember the hard work of the dev and try to find a donate link while i wait

jLynx, 2017-04-18:
I haven't had this issue for a long time, so I am going to close this

closed 2017-04-18 by jLynx