SpacehuhnTech/esp8266_deauther #175: Bug in Testing Version

schinfo, 2017-04-04:
Hello,

maybe I've found a bug in the new testing version.
I've enabled the display (SSD1306) and it works perfect.
But if I select some scanned APs for attack and start it, although unselected APs was kicked off.
Then I select only this AP and it wasn't kicked off.
So I believe, the selected APs are not this one, which where kicked off. 
Can you check it and, if it's a bug, fix it in the next version.

Best regards

spacehuhn, 2017-04-04:
No I can't find such a bug.
It's working fine for me.
You can look into the webinterface to see if the AP selection is the same as on the display.

schinfo, 2017-04-04:
Hello spacehuhn,

it's working fine, if you use the web-browser ;-)
I can see the full protocol in the webinterface (*** are the right APs, deleted for ... ;-) ).

getting attacks JSON...{ "aps": ["***","***"], "clients": ["FF:FF:FF:FF:FF:FF - BROADCAST"], "attacks": [{"name": "deauth","status": "40pkts/s","running": 1},{"name": "beacon (clone)","status": "ready","running": 0},{"name": "beacon (list)","status": "no SSID","running": 0},{"name": "probe request","status": "no SSID","running": 0}],"ssid": []}
done

But if I use the mini-display and the three buttons  this line is missing. I see only:
running deauth attack... done
running deauth attack... done
running deauth attack... done
running deauth attack... done
running deauth attack... done
... and so on

Although, if I use the web-interface, everything is shown on the mini-display too, but I can only scroll with up and down to "wifi on", "scan" and "start/stop attack".  The scanned APs are shown on the display but I can't scroll to this lines.
If I select scan on the mini-display with the select-button it's working again.

I hope this information can give you some hints about the problems.
And sorry, I'm a real Bavarian - so English isn't my mother-language ;-)

spacehuhn, 2017-04-07:
I updated the testing branch ;)

closed 2017-04-23 by spacehuhn