SpacehuhnTech/esp8266_deauther #176: "An error occurred while uploading the sketch"

arthurptj, 2017-04-06:
Windows 10 
arduino 1.8.2
esp8266 12e

`Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\alphabet\AppData\Local\Temp\arduino_cache_191943\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadTool_esptool,UploadSpeed_9600,FlashSize_4M3M_f8cacb6aa4e2fec0887666979d2d3248.a
Sketch uses 545674 bytes (52%) of program storage space. Maximum is 1044464 bytes.
Global variables use 56084 bytes (68%) of dynamic memory, leaving 25836 bytes for local variables. Maximum is 81920 bytes.
Uploading 549824 bytes from C:\Users\alphabet\AppData\Local\Temp\arduino_build_679688/esp8266_deauther.ino.bin to flash at 0x00000000
......................................................................................................................................................................................................................................................................An error occurred while uploading the sketc`

tobozo, 2017-04-07:
Sounds like a bad USB cable

spacehuhn, 2017-04-07:
We can't help you really here. 
`An error occurred while uploading the sketc` doesn't say anything.

haxord7, 2017-04-07:
use speed 115200 spped , at very fast i face same issue then i just change the speed and its ok  now.

Elora393, 2017-04-08:
same problem also mine

Sasco192, 2017-04-20:
@arthurptj Try installing the CP2102 driver on your PC, find it here: http://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

UberSlackr, 2017-04-21:
Getting the same error. Even with "show verbose output during: upload" it only displays

```
esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 9600
	setting port from COM1 to COM3
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat C:\Users\username\AppData\Local\Temp\arduino_build_688625/esp8266_deauther.ino.bin success
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_open
Uploading 551856 bytes from C:\Users\username\AppData\Local\Temp\arduino_build_688625/esp8266_deauther.ino.bin to flash at 0x00000000
	erasing flash
	size: 086bb0 address: 000000
	first_sector_index: 0
	total_sector_count: 135
	head_sector_count: 16
	adjusted_sector_count: 119
	adjusted_size: 077000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting serial port timeouts to 10000 ms
	setting serial port timeouts to 1000 ms
	espcomm_send_command: receiving 2 bytes of data
	writing flash
............................................................................................................................................................
..........................................................................................................An error occurred while uploading the sketch
```

jLynx, 2017-04-21:
Looks like you haven't extracted the zip file

spacehuhn, 2017-04-21:
Yes @arthurptj and @UberSlackr you have to extract the zip file first before you can compile it!

UberSlackr, 2017-04-22:
@jLynx @spacehuhn how would I have managed to do steps 9 through 15 if I had not already extracted the zip?

spacehuhn, 2017-04-22:
@UberSlackr 

> C:\Users\username\AppData\Local\Temp\arduino_build_688625/esp8266_deauther.ino.bin

I don't know what you did but I can see that you haven't extracted it.
Same for @arthurptj 

> C:\Users\alphabet\AppData\Local\Temp\arduino_build_679688/esp8266_deauther.ino

arthurptj, 2017-04-22:
@spacehuhn + @UberSlackr ,

I did extract ZIP, it was not a usb cable error, I ended up just moving on from arduino, and uploaded with esp lualoader. Probably was just impatience to try out all of the combinations of arduino settings.

closed 2017-04-22 by arthurptj

spacehuhn, 2017-04-22:
You guys should try to upload an example first. Because such upload errors are mostly not related to this project anyways.
Some knowledge and research skills are required.

It's just that we can't really help on such issues.
Good to hear that you managed to get it working 👍

TomiMandalaPutra, 2019-02-18:
Windows 7 64bit
IDE Arduino 1.8.9
ESP8266 (NodeMcu V3 Lolin CH30G)

Help me.... i'm newbie 

Sketch uses 258604 bytes (24%) of program storage space. Maximum is 1044464 bytes.
Global variables use 26696 bytes (32%) of dynamic memory, leaving 55224 bytes for local variables. Maximum is 81920 bytes.
C:\Users\Programmer\AppData\Local\Arduino15\packages\esp8266\tools\esptool\2.5.0-3-20ed2b9/esptool.exe -vv -cd nodemcu -cb 9600 -cp COM19 -ca 0x00000 -cf C:\Users\PROGRA~1\AppData\Local\Temp\arduino_build_664185/Blink.ino.bin 
esptool v0.4.13-1-gf80ae31 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to nodemcu
	setting baudrate from 115200 to 9600
	setting port from COM1 to COM19
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed