SpacehuhnTech/esp8266_deauther #178: Packets rate

haxord7, 2017-04-07:
In version 1.2 packet rate has lilmited only 20pkts if i change this to 50 or or more its continuesly sends 20pkts . version 1.1 has no problem like this . send upto 300+ pkts ,how to do same thing in version 1.2 ,??

i face another problem when first time i plugged and start attacking page is not responding but blue led blinking like its running ok but acctually i didn't acces the attack page .after 20/22 secend its running smoothly no problem .i face this stucking issue  when i plugged esp first time ,then 20/22 secend left i got the deauth page ,how to fix it??

please look at the picture 

---

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

---

spacehuhn, 2017-04-07:
It's both working fine for me using the latest testing branch.

haxord7, 2017-04-07:
thanks ,testing branch is running smoothly no problem . :+1:

closed 2017-04-07 by haxord7