SpacehuhnTech/esp8266_deauther #179: Some Question ?

Elora393, 2017-04-08:
I have 5 Esp8266-01 3 of esp i have used 2 more left ,i want to use it as deauther ,i have some  question 1) can i have upload source code via arduino ide  ??

i saw there some .bin version ,what firmware should i use ? 

can i select board Nodemcu-12 E and 3M spiff uploading time? for (esp-01)

how much highest pkts can send esp-deauther ?

(Sorry for this type of question) 
---

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

---

spacehuhn, 2017-04-08:
Everything is described in the readme.
Yes you can upload the bin file, you have to check if you need the 1mb or 512kb because there are ESP-1 s with both flash sizes.
And yes you can upload using Arduino then you have to set it to generic ESP8266 and give it the right flash size.

closed 2017-04-08 by spacehuhn

Elora393, 2017-04-08:
can you tell me what is the different between 1mb bin and 4mb bin  , seleccting generic esp 8266 not uploading