SpacehuhnTech/esp8266_deauther #185: No networks found

digounet, 2017-04-12:
Hi, I've succesffully flashed it into my ESP-01 board but no networks are listed. Any issue with the version?

jLynx, 2017-04-12:
I don't have this issue running  the latest version on both my boards, are you sure there are networks around you are are 2.4 ghz and not 5ghz?

digounet, 2017-04-12:
Hello,  I don't know why but if I connect to the 8266 using my notebook no networks are listed. But if I use my iphone to connect then it works but the deauth attack isn't working (maybe my AP isn't vulnerable). Thanks

spacehuhn, 2017-04-12:
Use an up to date version of Firefox or Chrome and the scan should work, unless there are no networks in range of course.

WillGreen98, 2017-04-12:
Could Tor potentially work? Scan for bridges etc...

closed 2017-04-20 by spacehuhn