SpacehuhnTech/esp8266_deauther #187: I F'ed up

djwildbill, 2017-04-14:
So i messed up i changed the password is there a way to hack it back to default
---

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

---

poslogic, 2017-04-14:
Use the settings page to change the password.

Erase the EEPROM and re-flashing to get a clean start.

djwildbill, 2017-04-14:
That was the issue I changed the password in the settings and my smartphone auto correct the darn thing... 

I tried refreshing it and every stated the same

spacehuhn, 2017-04-14:
Connect your dev board, open the serial monitor in arduino and click on reset on the board.
It will print out all settings including the password.

poslogic, 2017-04-14:
Or type the same thing into your phone again and see what it automatically corrects to?

But using serial monitor will show you the password :)

djwildbill, 2017-04-15:
I tried but I found out what happened when I did it and do the web interface they capitalize the first letter but when I go to type in the password for the Wi-Fi it isn't capital but thank you for let me know that cereal would let me know that information

closed 2017-04-15 by spacehuhn

RiccardoMax, 2017-04-21:
Hey guys, I got the same problem but when i click reset on the board it gives me this: 8P⸮t⸮⸮⸮⸮⸮Xp⸮⸮
Any Help?

spacehuhn, 2017-04-21:
You have to set the Baudrate to 115200 in the serial monitor.

djwildbill, 2017-04-21:
Thanks Spacehun this thing rocks

RiccardoMax, 2017-04-21:
Thanks spacehuhn, I managed to solve this issue by reflashing the unit and then reinstalling the software.