SpacehuhnTech/esp8266_deauther #188: Need ur help

J4ck02, 2017-04-15:
Hey Spacehuhn

i know it´s not a real issue but i don´t know how to contact you.
I´ve build an enclosure for my ESP8266. Now i would like to ask would it be possible to connect a extern Led to light up while an attack is active i don´t want to use the build in LED.
If this is the wrong place to ask please head me over to the right one.
Help would be much appreciated.
Thanks in advance J4ck

spacehuhn, 2017-04-15:
One way would be to connect the LED to pin 2 which is the pin for the built-in LED.
Otherwise you have to change the `Attack::refreshLed` function in `Attack.h` and change the `digitalWrite` to whatever pin you wanna use. You also have to change it in the `esp8266_deauther.ino` at `pinMode(2, OUTPUT)`.

J4ck02, 2017-04-15:
That sounds good so pin2 equals D2 ?

spacehuhn, 2017-04-15:
I don't know, you have to look that up for your board.

J4ck02, 2017-04-15:
Ok i´ll do that, thanks for your fast reply :)

spacehuhn, 2017-04-15:
Your welcome :)

closed 2017-04-15 by spacehuhn

J4ck02, 2017-04-15:
Sry man ik ur busy but it´s not working :/

poslogic, 2017-04-15:
It would help if you could identify what ESP board that you are using? 

Guessing that it's a WeMos D1 Mini?
Make sure that you use the 3.3v gpio not the 5v

Best to get a simple blink sketch working first before attempting to use with this project

J4ck02, 2017-04-15:
I got the blink sketch working man. Im using this Board [http://www.banggood.com/de/NodeMcu-Lua-WIFI-Internet-Things-Development-Board-Based-ESP8266-CP2102-Wireless-Module-p-1097112.html?rmmds=myorder](url)

poslogic, 2017-04-16:
The link does not work for us because it is linking into your orders from your account.

Did you get the blink sketch to work with an external led?

poslogic, 2017-04-16:
The link does not work for us because it is linking into your orders from your account.

Did you get the blink sketch to work with an external led?

But looks like you have a Node MCU Dev board?

J4ck02, 2017-04-16:
Yes it´s a NodeMCU sry for the corrupt link
 So i got the led kinda working but if i start the attack it turns off :D any ideas ?

J4ck02, 2017-04-16:
Fail alarm. Got the Blink working with D0 it´s blinking but if i do likespacehuhn sad 
```
One way would be to connect the LED to pin 2 which is the pin for the built-in LED.
Otherwise you have to change the Attack::refreshLed function in Attack.h and change the digitalWrite to whatever pin you wanna use. You also have to change it in the esp8266_deauther.ino at pinMode(2, OUTPUT).
```
it won´t light up the LED

J4ck02, 2017-04-16:
[image]

Works like a charm! Thank u guys ;)