SpacehuhnTech/esp8266_deauther #192: LCD and button wiring

ErfanDL, 2017-04-18:
Hi. I need LCD and button wiring

spacehuhn, 2017-04-18:
````#ifdef USE_DISPLAY

#include <Wire.h>

//include the library you need
#include "SSD1306.h"
//#include "SH1106.h"

//button pins
#define upBtn D6
#define downBtn D7
#define selectBtn D5

#define buttonDelay 180 //delay in ms

//render settings
#define fontSize 8
#define rowsPerSite 8

//create display(Adr, SDA-pin, SCL-pin)
SSD1306 display(0x3c, D2, D1);
//SH1106 display(0x3c, D2, D1);

int rows = 3;
int curRow = 0;
int sites = 1;
int curSite = 1;
int lrow = 0;
#endif```

closed 2017-04-18 by spacehuhn

ErfanDL, 2017-04-18:
is compiled BIN have LCD feature ?

schinfo, 2017-04-18:
Hello ErfanDL,

please use your brain, not your keyboard

ErfanDL, 2017-04-18:
Im wiring LCD but not worked

[image]

schinfo, 2017-04-18:
Connect the white wire to 3V3 (Pin left of the red wire)

ErfanDL, 2017-04-18:
@schinfo 
Im do it but not worked ! wiring SDA to D2 and SCL to D1 - VCC to 3.3 & GND to GND

schinfo, 2017-04-18:
You must use the Arduino-IDE, not the BIN-File

schinfo, 2017-04-18:
And read the hints in "esp8266_deauther.ino"

schinfo, 2017-04-18:
Line 16:  //#define USE_DISPLAY /* <-- uncomment that if you wanna use the display */

schinfo, 2017-04-18:
It's not enabled in the BIN-Files

spacehuhn, 2017-04-18:
Please don't spam.
It's for advanced users, you have to use some knowlege about how to use an OLED.
I can't explain every little detail.

ErfanDL, 2017-04-18:
Thanks. its worked with arduino ide

[image]