SpacehuhnTech/esp8266_deauther #193: Attack page continously refreshes about 5 times per second, unresponsive.

poorredneck, 2017-04-18:
Board: Wemos Mini D1 4MB (ESP12E based)
Latest master variant from Github. 
Arduino 1.8.1 on Windows 10
All how-to steps done according to readme.
Client: Two different android phones, both running android marshmallow.
Broswer: Chrome for Android & built in Samsung Browser and built in LG browser.

Issue:

I select an AP, and goto the attack page. The attack page just continuously refreshes over and over and over about 5 times per second. No APs listed on attack page. Can't press any buttons due to continuous refreshing. All other pages work fine. 

Selecting an AP on AP page, then going to clients page does succesfully scan for clients. Clients are shown. Goto attack page and attack page just keeps refreshing. Does not show selected client or selected AP.

Steps taken to try and resolve:
Thought maybe something in the EEPROM was corrupt. Downloaded the ESP Download tool, and completely wiped all 4MB of internal memory, using a 1MB "empty" bin file (all FF) from memory starting at address 0x000000. Verified EEPROM was wiped (all stored settings are back to default, SSID and password back to original settings.
Re-compiled sketch from latest download and uploaded.
Tried 2 different android based clients. Can not connect from my windows 10 computer, only mobile phones.

I can try and post a video but will wait unitl I get home to use my wife's phone to record with.

Thanks,
Jon aka poor_red_neck

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

---

poslogic, 2017-04-18:
It might help more if you could upload any output you get from the Arduino serial monitor to pastebin and link to it here.

f1vefour, 2017-04-20:
I get the exact same behavior.

This is the output of serial monitor:

https://pastebin.com/5NLYKa9K

soaron, 2017-04-28:
I have the same problem ('sending AP scan result JSON done' message) when using Wemos Mini Pro 16 (ESP-8266F based, CP2104). The same code runs perfectly on Wemos Mini D1 4MB (CH340G) using the same environment (Windows 7, Arduino IDE 1.6.11, Android 6.0 / Chrome).

spacehuhn, 2017-08-09:
This should be fixed by now.

closed 2017-08-09 by spacehuhn