SpacehuhnTech/esp8266_deauther #195: Arduino: 1.8.2 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"  sketch\Attack.cpp: In member function 'bool Attack::send()':  Attack.cpp:102: error: 'wifi_send_pkt_freedom' was not declared in this scope     if (wifi_send_pkt_freedom(packet, packetSize, 0) == -1) {                                                    ^  exit status 1 'wifi_send_pkt_freedom' was not declared in this scope  This report would have more information with "Show verbose output during compilation" option enabled in File -> Preferences.

helpme1234432, 2017-04-19:
---

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

---

jLynx, 2017-04-19:
Please don't post spam. If you have a problem, please format the issue correctly and ask a question

helpme1234432, 2017-04-19:
Arduino: 1.8.2 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"  sketch\Attack.cpp: In member function 'bool Attack::send()':  Attack.cpp:102: error: 'wifi_send_pkt_freedom' was not declared in this scope     if (wifi_send_pkt_freedom(packet, packetSize, 0) == -1) {                                                    ^  exit status 1 'wifi_send_pkt_freedom' was not declared in this scope  This report would have more information with "Show verbose output during compilation" option enabled in File -> Preferences.

helpme1234432, 2017-04-19:
The error im getting ^^^

jLynx, 2017-04-19:
Again, please format the issue properly and give us some background, or we cannot help you and your issue will be closed

SaltyViper, 2017-04-19:
You have not properly configured the arduino IDE as shown in [the instructions,](https://github.com/spacehuhn/esp8266_deauther/#compiling-the-source-with-arduino) Make sure you don't miss **any** steps.

SaltyViper, 2017-04-19:
@helpme1234432

helpme1234432, 2017-04-19:
is this video explain it right?  
https://www.youtube.com/watch?v=MOscKnm8IcY&t=261s

helpme1234432, 2017-04-19:
ive done this whole thing 3 times, the error code is poping up, plz help

closed 2017-04-19 by spacehuhn