SpacehuhnTech/esp8266_deauther #197: Problem when uploading...  Any help wanted!

DMCDBRE, 2017-04-19:
---
This is my error code:

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.
Arduino: 1.8.2 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

Build options changed, rebuilding all
Archiving built core (caching) in: /var/folders/3b/hw517cb95_l8hbsz1dw957_w0000gn/T/arduino_cache_514882/core/core_esp8266_esp8266_generic_CpuFrequency_80,FlashFreq_40,FlashMode_dio,UploadSpeed_115200,FlashSize_512K64,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____c894169e36ffd1bb6f262128486a9d64.a
/Users/dan/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: esp8266_deauther.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Thanks, 
Dan

---

closed 2017-04-19 by spacehuhn

DMCDBRE, 2017-04-19:
@spacehuhn How is this invalid? I'm asking for help. I dont know whats going wrong and need help.

jLynx, 2017-04-19:
You have not said what you are doing to get this error, what version you have installed and what ESP8266 Module you have

DMCDBRE, 2017-04-19:
Im using Arduino 1.8.2. Is that what your asking. Im sorry I'm new and thought I knew what i was doing but i guess not..